ANYbotics/grid_map #247: Error grid_map_pcl

brunoeducsantos, 2020-07-08:
Hi,
Thanks for your awesome package.
While trying to compiling the package, I had the following error:

``` 
/home/bruno/grid_ws/devel/lib/libgrid_map_pcl.so: undefined reference to `typeinfo for YAML::BadConversion'
/home/bruno/grid_ws/devel/lib/libgrid_map_pcl.so: undefined reference to `YAML::BadConversion::~BadConversion()'
/home/bruno/grid_ws/devel/lib/libgrid_map_pcl.so: undefined reference to `vtable for YAML::BadConversion'
/home/bruno/grid_ws/devel/lib/libgrid_map_pcl.so: undefined reference to `pcl::search::Search<pcl::PointXYZ>::getName[abi:cxx11]() const'

``` 

Any help would be appreciated.

Regards,
Bruno

maximilianwulf, 2020-07-08:
Hey @brunoeducsantos, which branch are you using?

brunoeducsantos, 2020-07-08:
Master

brunoeducsantos, 2020-07-08:
I tried this on `GridMapPclLoader` ,
```
#include <pcl/search/impl/search.hpp>
#ifndef PCL_NO_PRECOMPILE
#include <pcl/impl/instantiate.hpp>
#include <pcl/point_types.h>
PCL_INSTANTIATE(Search, PCL_POINT_TYPES)
```
And it sorted the pcl issue

maximilianwulf, 2020-07-08:
Hm interesting, so it is solved?

brunoeducsantos, 2020-07-08:
The Yaml issue hasn't been solved, but probably it is a `yaml-cpp` version issue.
So, i guess i will close the issue for now.

closed 2020-07-08 by brunoeducsantos

EricVoll, 2021-09-30:
@brunoeducsantos have you ever figured out which yaml-cpp version to use? I am facing the same issue. I've tried 0.6.0, 0.3.0 and 0.2.7 without success so far.