ANYbotics/grid_map #252: cmake_code_coverage is not a valid rosdep key

StefanFabian, 2020-08-10:
https://github.com/ANYbotics/grid_map/blob/270a9cc7f4551fc4b4488a5d3a8c98f7c42b9269/grid_map_core/package.xml#L14

This recent change broke our private build server bloom builds since `cmake_code_coverage` is not a known rosdep key.
I'm also not sure what project that is referring to as I couldn't find any project named `cmake_code_coverage` on github or google.
There is a [`code_coverage`](https://index.ros.org/r/code_coverage/github-mikeferguson-code_coverage/) project which is also available using rosdep but judging from the `CMakeLists.txt` that's not the one used.

If whatever you're using is a system package, it would be great if you could add it to the [rosdep database](https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml) (you would have to change the key to the debian package name though).
If it is not, it would be great if you could post a link to the project, so I can fix the builds on our server without having to maintain a fork :)

YoshuaNava, 2020-08-11:
Hi @StefanFabian,
Thank you for reporting this issue. It has been fixed on master.

Best,
Yoshua Nava

StefanFabian, 2020-08-12:
Thank you for the quick fix!

closed 2020-08-12 by StefanFabian