ANYbotics/grid_map #256: Question # Tests Compiled Wrong

tuzhiming2016, 2020-08-29:
I found that the three files have some problem.
    CubicConvolutionInterpolationTest.cpp \
    CubicInterpolationTest.cpp \
    EigenPluginsTest.cpp

first:
in the file #include "grid_map_core/TypeDefs.hpp"
enum class InterpolationMethods has no member called INTER_CUBIC and INTER_CUBIC_CONVOLUTION.
Are these your TODOs? So The two tests may have some wrongs if compiled?
TODOs{gm::InterpolationMethods::INTER_CUBIC
	  gm::InterpolationMethods::INTER_CUBIC_CONVOLUTION}????

second:
in the file EigenPluginsTest.cpp
Eigen::Matrix has no member named ‘numberOfFinites’ as well as sumOfFinites minCoeffOfFinites and maxCoeffOfFinites.
I found that the definition is in grid_map_core/eigen_plugins/DenseBasePlugin.hpp, however there is no one file include that, and there would be wrong if compiled.
Is this also your TODO??

Or if not , Can you give me some solutions to that? And I am using the package.
Thanks a lot~

TuZhiming

maximilianwulf, 2020-09-03:
Hey @tuzhiming2016, it looks like you found some issues in our tests. Could you submit a PR to solve those?

CxyCH, 2021-04-14:
Hi  @tuzhiming2016 :
    I have found this problem too. Do you know how to fix it now?