ANYbotics/grid_map #261: grid map visualization & convert to image

kwangwookyun, 2020-09-17:
I want the following 2 things.

1. I want to convert the result of the visualization in rviz with color to an image (figure 1 in Rviz to image data in ROS)
2. I want to merge it with the camera image. So, I need to convert from visualization result in rviz to CV type.

[image]
figure 1.  the result of the visualization in rviz with color

Currently I use the grid_map_cv package to convert the gridmap to cv, but it is different from the result of being created as a black and white image and visualrized in rviz.

Is there any way I can do the 2 things I want? Please Any help would be appreciated.

maximilianwulf, 2020-09-25:
Hey @kwangwookyun ,

> I want to convert the result of the visualization in rviz with color to an image (figure 1 in Rviz to image data in ROS)

Yes, use the opencv function `applyColorMap`. You can use the technique described in this thread: https://stackoverflow.com/questions/13840013/opencv-how-to-visualize-a-depth-image

> I want to merge it with the camera image. So, I need to convert from visualization result in rviz to CV type.

Can you elaborate on this idea?

kwangwookyun, 2020-10-01:
> Hey @kwangwookyun ,
>
>
> Yes, use the opencv function `applyColorMap`. You can use the technique described in this thread: [...]

@maximilianwulf
Thanks for answering.

- merging grid map image with camera image means overlay .

The grid map data type is visualized through rviz as shown in figure 1.

I want to convert the result visualized in rviz into an image and "overlap" it with the camera image.
This will be shown in a form similar to AR(augmented reality).

As a result, i want to visualize it as follows. (Figure 3)

Figure 2. Before to merge
[image]

Figure 3. result overlay gridmap to camera image
[image]

In order to look like the above, the rviz visualization result must be converted into an image. I want to do this.
The opencv_demo package converts the gridmap to an image, but it is not visualized like rviz

maximilianwulf, 2020-10-06:
Hey @kwangwookyun, I think RVIZ can generate this for you, right? 

So if you choose 
[image]

you get this description

[image]

and the image will render everything as an image what is currently visible. 

[image]

> The opencv_demo package converts the gridmap to an image, but it is not visualized like rviz

The output of the grid_map_cv is an image that is seen from above the grid map. If you want to overlay it on a 3D scene you would have to do the projections by yourself.

kwangwookyun, 2020-10-23:
thank you! I solved the issue.

closed 2020-10-23 by kwangwookyun