ANYbotics/grid_map #265: Compile error help needed...

thezest76, 2020-10-14:
Hello,

While I was trying to compile the package, I met the error below. 
Could you tell me how to fix this problem ?

My system is Ubuntu 2020 + ROS noetic

Best,

> 
> In file included from /opt/ros/noetic/include/rviz/message_filter_display.h:37,
>                  from /home/hankm/catkin_ws/src/grid_map/grid_map_rviz_plugin/include/grid_map_rviz_plugin/modified/message_filter_display.h:13,
>                  from /home/hankm/catkin_ws/build/grid_map/grid_map_rviz_plugin/include/grid_map_rviz_plugin/../../../../../src/grid_map/grid_map_rviz_plugin/include/grid_map_rviz_plugin/GridMapDisplay.hpp:15,
>                  from /home/hankm/catkin_ws/build/grid_map/grid_map_rviz_plugin/include/grid_map_rviz_plugin/moc_GridMapDisplay.cpp:9:
> /opt/ros/noetic/include/rviz/display_context.h: In member function ‘std::shared_ptr<tf2_ros::Buffer> rviz::DisplayContext::getTF2BufferPtr() const’:
> /opt/ros/noetic/include/rviz/display_context.h:100:31: error: ‘class rviz::FrameManager’ has no member named ‘getTF2BufferPtr’
>   100 |     return getFrameManager()->getTF2BufferPtr();
>

maximilianwulf, 2020-10-14:
Hey @thezest76, the library is currently not upgraded to Noetic, this will happen soon. In the meantime, you can check out the branch from this MR to compile it. https://github.com/ANYbotics/grid_map/pull/231

thezest76, 2020-10-14:
Hello,

Thanks for the quick response.
Could you please tell what would be the branch name ?
Are we talking about bf0be3a ?  How do I check-out this commit from my
local directory ?

Best,

maximilianwulf, 2020-10-14:
It is from a forked repo: https://github.com/jspricke/grid_map/tree/noetic

You can read more about git on this blog post: https://www.tomasbeuzen.com/post/git-fork-branch-pull/

thezest76, 2020-10-15:
Thanks for the kind explanation.
It is working now.

Best,

maximilianwulf, 2020-10-15:
Glad to hear. Thank you for using grid map.

closed 2020-10-15 by maximilianwulf

liuzm-slam, 2020-12-24:
how do you deal with this problem? i have the same problem, i need your help

maximilianwulf, 2021-01-06:
Hi @liuzm-slam can you give a bit more context, which version of OS are you using and which ROS version? 

Can you also add your specific error?