ANYbotics/grid_map #267: MessageFilterDisplay Crash

StefanFabian, 2020-10-22:
After updating, I'm getting the following crash when starting RViz with a config that contains a GridMap display.

Stacktrace says the relevant line is:
https://github.com/ANYbotics/grid_map/blob/b4376234e7a9edadca1bb1302a4b27f2374637e3/grid_map_rviz_plugin/include/grid_map_rviz_plugin/modified/message_filter_display.h#L105

If I replace that call with a fixed number, e.g. 100, it crashes at the access of `queue_size_property_` in `subscribe`.
If I replace that, too, everything works fine.
I don't have a clue why `queue_size_property_` is invalid, though. The debugger won't show any member of `_RosTopicDisplay` maybe that is related.

maximilianwulf, 2020-10-23:
Hey @StefanFabian, we did not observe that issue yet. 

Can you share a core dump?

maximilianwulf, 2020-10-28:
Can you share your ROS version?

StefanFabian, 2020-10-28:
Sorry for the late reply, I was occupied with a paper deadline.
I'm on Melodic.
I'll check if I can figure out what's happening or get a core dump in the next few days.

Austin-Hu, 2020-11-09:
Hi @StefanFabian & @maximilianwulf,

Please refer to my [comment](https://github.com/ANYbotics/elevation_mapping/issues/141#issuecomment-724272147) in `elevation_mapping` Issue #[141](https://github.com/ANYbotics/elevation_mapping/issues/141) which was closed.

Thanks!

maximilianwulf, 2020-11-11:
Can someone check if rviz changed this property recently?