ANYbotics/grid_map #271: 2D Grid Map projection to an image

brunoeducsantos, 2020-11-20:
Hi,

I was trying to project a grid map keeping the colors to a image like this:

`
cv::Mat map;
	GridMapCvConverter::toImage<unsigned char,3>(gridMap, layer, CV_16UC3,map);
`

Although, I am getting a mix of gray and yellow colors.

Could you help me with this issue?

Thanks,
Bruno

maximilianwulf, 2020-11-30:
Hey @brunoeducsantos, the function does not directly generate a colorful image. Your output is expected. 

In case you want a color map, you have to apply a color map to the image. 

This snippet should roughly do what you want: 

```
       // Holds the colormap version of the image:
        Mat cm_img0;
        // Apply the colormap:
        applyColorMap(img0, cm_img0, COLORMAP_JET);
        // Show the result:
        imshow("cm_img0", cm_img0);
        waitKey(0);
```

Inspired by this stackoverflow thread: https://stackoverflow.com/questions/13840013/opencv-how-to-visualize-a-depth-image

Hope that answers your question.

brunoeducsantos, 2020-12-01:
Thanks for your hint. I will try it out and let you know.

brunoeducsantos, 2020-12-03:
@maximilianwulf  It worked like a charm. Thanks a lot!

closed 2020-12-03 by brunoeducsantos

maximilianwulf, 2020-12-04:
Glad to hear.