ANYbotics/grid_map #272: getIndex() Unable to Obtain Index for Maximum Negative Length of Map

BruceChanJianLe, 2020-11-23:
My grid_map is geometry size is as below:

```cpp
map_.setGeometry(grid_map::Length(1.0, 1.0), 0.05, grid_map::Position(0.0, 0.0));
```

When my position is -0.5 (return from getClosestPositionInMap()), which is still valid in the map, I get the error of 
```bash
/usr/include/eigen3/Eigen/src/Core/DenseCoeffsBase.h:365: Eigen::DenseCoeffsBase<Derived, 1>::Scalar& Eigen::DenseCoeffsBase<Derived, 1>::operator()(Eigen::Index, Eigen::Index) [with Derived = Eigen::Matrix<float, -1, -1>; Eigen::DenseCoeffsBase<Derived, 1>::Scalar = float; Eigen::Index = long int]: Assertion `row >= 0 && row < rows() && col >= 0 && col < cols()' failed.
```
However, if my position is `0.5` (return from getClosestPositionInMap()), I do not get the error, and I am able to obtain the index.

I specifically mentioned getClosestPositionInMap() function as I am not sure whether should the getClosestPositionInMap() return something like `-0.49` instead of `-0.5`, or is this a bug inside of getIndex().

Below code will fail when any of the x or y position is `-0.5` which is the maximum negative length of the map.
```cpp
if(map_.getIndex(points_["point_a"], index_a_)
&& map_.getIndex(points_["point_b"], index_b_))
 {
    // Loop through line region
    for(grid_map::LineIterator itr(map_, index_a_, index_b_); !itr.isPastEnd(); ++itr)
    {
        // Change value from 0.0 to 1.0
        map_.at("type", *itr) = 1.0;
        // Publish map
        publish();
    }
}
```

Thank you for taking your time to look into this matter:D

maximilianwulf, 2020-11-30:
Hey @BruceChanJianLe, thank you for submitting the bug. Could you submit a short unit test  into this file: https://github.com/ANYbotics/grid_map/blob/master/grid_map_core/test/GridMapTest.cpp. I guess you can copy the code from above.

BruceChanJianLe, 2020-12-02:
Hi @maximilianwulf, thanks for getting back. I have submitted a short unit test as requested in https://github.com/ANYbotics/grid_map/pull/275. Please have a look at the failing test `NegativeClosestPositionInMap`. Thank you:D

maximilianwulf, 2020-12-02:
Thank you, we will have a look at it.

BruceChanJianLe, 2020-12-02:
You're welcome! Thank you for taking your time to look into this issue.

maximilianwulf, 2020-12-08:
Hey @BruceChanJianLe, indeed the function `getClosestPositionInMap` gives values at the border of the map, which can in some cases lead to indices outside of the map. I made a fix that should make the function save. It will be merged in the coming days. Thank you for reporting the issue.

BruceChanJianLe, 2020-12-09:
Hi @maximilianwulf, thank you for looking into this issue, appreciate your time and contribution to the community. Please remember to close this issue when the fix is merged, before that, I shall leave this issue as open. Thanks again and take care!

maximilianwulf, 2020-12-16:
Hey @BruceChanJianLe, a fix is on the way. Let me know if you have further issues.

closed 2020-12-16 by maximilianwulf

BruceChanJianLe, 2020-12-28:
Hi @maximilianwulf, thanks for looking into this is issue. I will be looking forward to the fix! Thank you:)

maximilianwulf, 2020-12-29:
You can check out this commit https://github.com/ANYbotics/grid_map/commit/f1344bd946967d2806dc9d0e41b119e580398724

BruceChanJianLe, 2021-01-04:
Wonderful! I will test on it when it is successfully merged into master branch. Thank you @maximilianwulf !