ANYbotics/grid_map #274: Gridmap intitialization with interpolation

evenlund, 2020-11-30:
It appears that Gridmap is using linear interpolation between cells as default. The demo-example shows how to transfer one map to another using interpolators, but is it possible to set a default interpolator when initializing the gridmap? 

The example seems quite complicated using a yaml. Would it be possible to do interpolation directly in code?

maximilianwulf, 2020-11-30:
Hey @evenlund, check out this function: https://github.com/ANYbotics/grid_map/blob/d9ee02deb5af682339610ad2313c0bd9c052df75/grid_map_demos/src/InterpolationDemo.cpp#L268

You can use the example to craft your solution. 

Using yaml should not be complicated, it just sets one variable, check here: https://github.com/ANYbotics/grid_map/blob/d9ee02deb5af682339610ad2313c0bd9c052df75/grid_map_demos/src/InterpolationDemo.cpp#L245

evenlund, 2020-12-01:
It is sort of answering my question. I assume there is no constructor or init function that will set the interpolation used directly when creating the gridmap object, so you would need the original map and a duplicated map that has the interpolation you want.

What confuses me the most with the example is that there is no main function. I understand that this is not a forum, but could you tell me how to "convert" a standard ROS cpp file to one that accepts yaml input? I looked at the ROS wiki for "Parameter Server", but could not find info there.

maximilianwulf, 2020-12-01:
> It is sort of answering my question. I assume there is no constructor or init function that will set the interpolation used directly when creating the gridmap object, so you would need the original map and a duplicated map that has the interpolation you want.

Yes, I think so. 

[Here](https://github.com/ANYbotics/grid_map/blob/d9ee02deb5af682339610ad2313c0bd9c052df75/grid_map_demos/src/interpolation_demo_node.cpp) is the main function of the demo. 

The yaml file is read in the launch file at [this line](https://github.com/ANYbotics/grid_map/blob/d9ee02deb5af682339610ad2313c0bd9c052df75/grid_map_demos/launch/interpolation_demo.launch#L8).  From then on all the parameters in the file are available through the rosparam. The usage can be seen in https://github.com/ANYbotics/grid_map/blob/d9ee02deb5af682339610ad2313c0bd9c052df75/grid_map_demos/src/InterpolationDemo.cpp#L268

evenlund, 2020-12-10:
I think I got the interpolation working now. Unfortunately it does not change the representation I get of the data.

To explain a bit further I would like to place sensor data onto a flat gridded map similar to a checkered chess board, where each square has uniform color and no interpolation happens between the squares or internally in the square. It is perhaps more comparable to a rasterized image. I sort of obtain this by using a gridmap with flat height transformer, but I am trying to remove the interpolation that happens internally in the square and between the squares (see attachment). If I don't initialize the grid with a value when using "gridmap.add" I see that tiangles appear when data is added to a new position and is not populating the whole square.

- Is it possible to remove internal and external interpolation fully (preferrably on the original grid instead on a new gridmap)? Is there some kind of subgrid in each square?
- The coloring becomes a bit psycadelic to my taste with rainbow and two colors are a bit limiting. Is it possible to use a multi color map like in matplotlib, matlab, paraview and others with grid_map?
- Can you explain some more the difference between the demo variables "groundTruthResolution", "dataResolution", "interpolatedResolution" in reference to these questions?

[image]