ANYbotics/grid_map #276: Filter Demo - MockFilter errors

aled96, 2020-12-02:
Hi,

I tried to launch `roslaunch grid_map_demos filters_demo.launch` but I obtain the following error (and in RViz I cannot see anything:

MockFilter did not find parameter 'print_name'.
Could not configure the filter chain!

I tried to set print_name as a parameter, setting it so a string, but it does not work.

Thank you for the help

maximilianwulf, 2020-12-09:
Thank you for reporting, we will fix it.

maximilianwulf, 2020-12-10:
Should be fixed.

closed 2020-12-10 by maximilianwulf

jcnorby, 2020-12-16:
@maximilianwulf I'm hving a similar experience to @aled96 - the PR for this issue cleared the error, but something is still off in the demos that causes the filters to not show up. It looks as if the structure for the filter chain parameters changed at some point but the .yaml for the filter demos has not been updated. Any update on this would be much appreciated!

jcnorby, 2020-12-16:
I think this may just be a compiling issue. After some digging, I found that the filters demo seemed to just take forever in the MeanInRadiusFilter, which was surprising since that filter hasn't been modified in a few years. This prompted me to compile in Release mode (`catkin_make -DCMAKE_BUILD_TYPE=Release`), which worked like a charm. Everything is as expected now.

maximilianwulf, 2021-01-15:
@jcnorby Thank you for the update. Grid map indeed requires to be build in Release.