ANYbotics/grid_map #278: Grid map demo failing for interpolation_demo.launch

dlee640, 2020-12-13:
**I was able to run all the demos leading upto the interpolation_demo.launch. However, it is failing to run the demo for interpolation with following error and I don't understand how to fix the issue.**

I am getting exception:

```
RLException: [interpolation_demo.launch] is neither a launch file in package [grid_map_demos] nor is [grid_map_demos] a launch file name
The traceback for the exception was written to the log file
```
when I try to run
`roslaunch grid_map_demos interpolation_demo.launch`

I have used the following setup procedure. I couldn't use catkin_make because error occured with costmap2d, and when I used catkin build grid_map, it compiled fine.:

1. check env
printenv | grep ROS

2. source correct ros distribution
source /opt/ros/melodic/setup.bash

3. setup catkin workspace
mkdir -p ~/catkin_ws/src
cd ~/catkin_ws/src

4. install & clone prereq
sudo apt-get install libeigen3-dev
git clone https://github.com/ANYbotics/grid_map.git

5. build catkin workspace
catkin build grid_map

6. run unit tests
catkin build grid_map --no-deps --verbose --catkin-make-args run_tests

7. install rospkg 
pip install rospkg
pip install defusedxml

8. run demo
roslaunch grid_map_demos simple_demo.launch

jcnorby, 2020-12-16:
Not a dev, but it looks like the issue is that you haven't sourced your catkin workspace. It seems that installing grid_map via apt will put a number of your demos into your install space (`/opt/ros/melodic/share/grid_map_demos`), which ROS knows where to find because you sourced it in step 2. It looks like interpolation_demo.launch is NOT included in those, which is why some of the others work and that one doesn't. Try running `source ~/catkin_ws/devel/setup.bash` and see if that helps.

On a related note, I just solved that costmap2d issue - it stems from a missing package that typically ships with the navigation package. I think only full ROS installs include this (not desktop or base), and it seems grid_map doesn't include it as a dependency. Try `sudo apt install ros-melodic-navigation` then recompile.

maximilianwulf, 2021-01-15:
@dlee640 can you verify that everything is working for you? 

@jcnorby thank you for the answer :)

dlee640, 2021-01-28:
Sorry for the late reply. This solution worked!

closed 2021-01-28 by dlee640