ANYbotics/grid_map #279: error about grid map rviz plugin

liuzm-slam, 2020-12-25:
Errors     << grid_map_rviz_plugin:make /home/cnimi/segmap_ws/logs/grid_map_rviz_plugin/build.make.001.log
In file included from /home/cnimi/segmap_ws/build/grid_map_rviz_plugin/include/grid_map_rviz_plugin/modified/moc_message_filter_display.cpp:9:0:
/home/cnimi/segmap_ws/build/grid_map_rviz_plugin/include/grid_map_rviz_plugin/modified/../../../../../src/grid_map/grid_map_rviz_plugin/include/grid_map_rviz_plugin/modified/message_filter_display.h: In member function ‘void grid_map_rviz_plugin::MessageFilterDisplay<MessageType>::onInitialize()’:
/home/cnimi/segmap_ws/build/grid_map_rviz_plugin/include/grid_map_rviz_plugin/modified/../../../../../src/grid_map/grid_map_rviz_plugin/include/grid_map_rviz_plugin/modified/message_filter_display.h:84:69: error: ‘class rviz::DisplayContext’ has no member named ‘getTF2BufferPtr’
     tf_filter_ = new tf2_ros::MessageFilter<MessageType>(*context_->getTF2BufferPtr(), fixed_frame_.toStdString(), 1u, update_nh_);
                                                                     ^
make[2]: *** [CMakeFiles/grid_map_rviz_plugin.dir/include/grid_map_rviz_plugin/modified/moc_message_filter_display.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/cnimi/segmap_ws/build/grid_map_rviz_plugin/include/grid_map_rviz_plugin/../../../../src/grid_map/grid_map_rviz_plugin/include/grid_map_rviz_plugin/GridMapDisplay.hpp:16:0,
                 from /home/cnimi/segmap_ws/build/grid_map_rviz_plugin/include/grid_map_rviz_plugin/moc_GridMapDisplay.cpp:9:
/home/cnimi/segmap_ws/src/grid_map/grid_map_rviz_plugin/include/grid_map_rviz_plugin/modified/message_filter_display.h: In member function ‘void grid_map_rviz_plugin::MessageFilterDisplay<MessageType>::onInitialize()’:
/home/cnimi/segmap_ws/src/grid_map/grid_map_rviz_plugin/include/grid_map_rviz_plugin/modified/message_filter_display.h:84:69: error: ‘class rviz::DisplayContext’ has no member named ‘getTF2BufferPtr’
     tf_filter_ = new tf2_ros::MessageFilter<MessageType>(*context_->getTF2BufferPtr(), fixed_frame_.toStdString(), 1u, update_nh_);
                                                                     ^
In file included from /home/cnimi/segmap_ws/src/grid_map/grid_map_rviz_plugin/include/grid_map_rviz_plugin/GridMapDisplay.hpp:16:0,
                 from /home/cnimi/segmap_ws/src/grid_map/grid_map_rviz_plugin/src/GridMapDisplay.cpp:10:
/home/cnimi/segmap_ws/src/grid_map/grid_map_rviz_plugin/include/grid_map_rviz_plugin/modified/message_filter_display.h: In member function ‘void grid_map_rviz_plugin::MessageFilterDisplay<MessageType>::onInitialize()’:
/home/cnimi/segmap_ws/src/grid_map/grid_map_rviz_plugin/include/grid_map_rviz_plugin/modified/message_filter_display.h:84:69: error: ‘class rviz::DisplayContext’ has no member named ‘getTF2BufferPtr’
     tf_filter_ = new tf2_ros::MessageFilter<MessageType>(*context_->getTF2BufferPtr(), fixed_frame_.toStdString(), 1u, update_nh_);
                                                                     ^
make[2]: *** [CMakeFiles/grid_map_rviz_plugin.dir/include/grid_map_rviz_plugin/moc_GridMapDisplay.cpp.o] Error 1
make[2]: *** [CMakeFiles/grid_map_rviz_plugin.dir/src/GridMapDisplay.cpp.o] Error 1
make[1]: *** [CMakeFiles/grid_map_rviz_plugin.dir/all] Error 2
make: *** [all] Error 2

problem:
grid_map_rviz_plugin/include/grid_map_rviz_plugin/modified/message_filter_display.h:84:69: error: ‘class rviz::DisplayContext’ has no member named ‘getTF2BufferPtr’
     tf_filter_ = new tf2_ros::MessageFilter<MessageType>(*context_->getTF2BufferPtr(), fixed_frame_.toStdString(), 1u, update_nh_); 
getTF2BufferPtr()?

maximilianwulf, 2021-01-06:
Hey @liuzm-slam, it seems that you might have a wrong tf version locally installed. Which tf version are you using?

zlg9folira, 2021-02-01:
Same issue here, I get the following error message on building some of the packages:
```
error: ‘class rviz::DisplayContext’ has no member named ‘getTF2BufferPtr’
     tf_filter_ = new tf2_ros::MessageFilter<MessageType>(*context_->getTF2BufferPtr(), fixed_frame_.toStdString(), 1u, update_nh_);
```

I cloned the repository by `https://github.com/ANYbotics/grid_map.git` which sounds a fit for ROS melodic. Melodic has support for `getTF2BufferPtr` ([here](http://docs.ros.org/en/melodic/api/rviz/html/c++/classrviz_1_1DisplayContext.html)) , but Kinetic doesn't ([here](http://docs.ros.org/en/kinetic/api/rviz/html/c++/classrviz_1_1DisplayContext.html)). Is there any branch or remedy for Kinetic ?

liuzm-slam, 2021-02-01:
> Same issue here, I get the following error message on building some of the packages:
> 
> ```
> error: ‘class rviz::DisplayContext’ has no member named ‘getTF2BufferPtr’
>      tf_filter_ = new tf2_ros::MessageFilter<MessageType>(*context_->getTF2BufferPtr(), fixed_frame_.toStdString(), 1u, update_nh_);
> ```
> 
> I cloned the repository `https://github.com/ANYbotics/grid_map.git` which sounds a fit for ROS melodic. Melodic has support for `getTF2BufferPtr` ([here](http://docs.ros.org/en/melodic/api/rviz/html/c++/classrviz_1_1DisplayContext.html)) , but Kinetic doesn't ([here](http://docs.ros.org/en/kinetic/api/rviz/html/c++/classrviz_1_1DisplayContext.html)). Is there any branch or remedy for Kinetic ?

I have solve this problem, i download a new package, the package is low version，i download 1.6? for grid map(ros kinetic, ubuntu16.04). you can give me your email address, i will give you for my package of grid_map.

zlg9folira, 2021-02-01:
@liuzm-slam Thanks, the 1.6 worked though it is old and my question now is what will I miss in features if I rely on version 1.6 instead of 1.6.4 which has recently merged ? Does 1.6 have all the features as the new release ?

liuzm-slam, 2021-02-01:
> @liuzm-slam Thanks, the 1.6 worked though it is old and my question now is what will I miss in features if I rely on version 1.6 instead of 1.6.4 which has recently merged ? Does 1.6 have all the features as the new release ?

I use grid map in elevation mapping, and i dont know other features. sorry.

maximilianwulf, 2021-02-08:
Hey all, it seems this is a kinetic related issue. As @zlg9folira mentions one can use an older version of grid map. It should contain the core features, except some recent fixes. But in my view the software is stable.

Afiq11, 2021-05-25:
hi guys,, can i know where to get the older version?? for ros kinetic

maximilianwulf, 2021-05-31:
Hi @Afiq11 you can check out the release 1.6.

lei0907, 2021-09-04:
I use ROS Kinetic and Ubuntu 16.04.Now i face the same issue.Could you give me the low version.thanks

hanlinhung, 2021-12-14:
> I use ROS Kinetic and Ubuntu 16.04.Now i face the same issue.Could you give me the low version.thanks

we are on master branch by default, I've tried "git checkout 1.6.2" and it worked

zlg9folira, 2022-01-13:
@maximilianwulf, following up on this issue; versions 1.6.0 and 1.6.2 both work just fine on Kinetic. However, some useful features/examples are added to 1.6.4 including "interpolations". This version doesn't compile on Kinetic. Could anyone work around new features for Kinetic?

maximilianwulf, 2022-02-01:
@zlg9folira if there is enough request we can start a `kinetic-devel` branch that is maintained by the community, would you be up for contributing to it?