ANYbotics/grid_map #280: Grid Map to Gazebo

jcnorby, 2020-12-28:
As far as I can tell there isn't a straightforward way to export a grid map for use in Gazebo. Is this correct, and if so is there a recommended workaround such as a way to load a grid map from a stl file? My best guess is to convert an stl to an image file then load from there, but that would involve setting some extra parameters to make sure the heights were correctly preserved.

maximilianwulf, 2021-01-04:
Hi @jcnorby, yes, you are right. This functionality is not supported at the moment. Feel free to submit a PR if you found a good solution. 

One pointer I could give is the 3D visualization for RVIZ. In there we iterate through the grid map and create an ogre object. Maybe there are converters between ogre and stl. Check out this [function](https://github.com/ANYbotics/grid_map/blob/master/grid_map_cv/include/grid_map_cv/GridMapCvProcessing.hpp#L45).

micco00x, 2022-02-17:
I was looking for a solution a few months ago and I've just learned about https://github.com/fmrico/gazebo_gridmap_plugin. Maybe it could be a nice starting point.

maximilianwulf, 2022-05-03:
@micco00x wow, I was not aware of it. Thank you.