ANYbotics/grid_map #282: Using grid_map_core without ROS?

duncanrhamill, 2021-01-14:
Hi, I'm interested in using grid_map but without using ROS. I notice that grid_map_core is only dependent on Eigen, which is great, but it would be nice to have a true standalone version which can be built and installed as a shared lib. Are there any plans to support this?

If not would you accept PRs to enable this?

maximilianwulf, 2021-01-14:
Hi @duncanrhamill, thank you for the request. We don't plan to have a non-catkinized version. That is why we would not accept such an MR. 

But you can always ask for help here if you run into issues when porting the library.

closed 2021-01-14 by maximilianwulf

duncanrhamill, 2021-01-14:
Ok thanks, I understand. Do you have any objections to a public fork of at least grid_map_core and grid_map_pcl?

maximilianwulf, 2021-01-14:
No, not that I know of. I think that is fine, thank you for asking :)

But I guess we will not provide directly support for the fork.

duncanrhamill, 2021-01-14:
Yeah that's fair enough, thanks!