ANYbotics/grid_map #283: ROS2 port follow up

Marwan99, 2021-01-14:
- octomap_to_gridmap  demo node and launch files have been ported, however it has not been tested as `octomap_server`  has not been ported yet.
- Once `pcl_ros` binaries are available, grid_map should be ready to release.

SteveMacenski, 2021-01-15:
Renaming just to be a little more descriptive for passersby

maximilianwulf, 2021-01-15:
Thank you for the issue.

msy22, 2021-05-06:
How usable is the Foxy/gridmap port in it's current state? It appears that the `grid_map_sdf` and `grid_map_pcl` modules both depend on `pcl_ros`. But if neither of those are required, are all the other modules fully functional?

SteveMacenski, 2021-05-06:
Yes, that is my understanding as well! You can definitely use it in Foxy and if you run into any issues, let us know!

SteveMacenski, 2021-05-13:
There's a PR resolving that last item, so I think that means we can release binaries @Marwan99 @maximilianwulf ?

Marwan99, 2021-05-16:
Can binaries be released before the first item on the list is resolved?

> octomap_to_gridmap demo node and launch files have been ported, however it has not been tested as octomap_server has not been ported yet.

FYI, there has been some movement on the octomap_mapping [port](https://github.com/OctoMap/octomap_mapping/issues/76) lately.

SteveMacenski, 2021-05-17:
What's the scope of the octomap stuff? Do we actually need octomap headers for something or just launching in a launch file? Could that one be temporarily disabled until ported?

SteveMacenski, 2021-05-20:
To release to galactic (or any ROS2) we need the filters to be exporting its dependencies https://github.com/ros/filters/pull/50

Marwan99, 2021-05-27:
> What's the scope of the octomap 

I think this [launch file](https://github.com/ANYbotics/grid_map/blob/c5d985a5f7a9ce3a5a944a74164270f01ae24996/grid_map_demos/launch/octomap_to_gridmap_demo_launch.py#L35-L43) is the only thing that needs octomap_mapping, its already disabled there.

SteveMacenski, 2021-05-27:
OK, I think  filters is the only blocker then for attempting a release

maximilianwulf, 2021-05-31:
Hey, releasing binaries definitely sounds great. How does it work in the ROS2 world?

SteveMacenski, 2021-06-01:
It is the same method as in ROS1, PR against rosdistro against the distribution of choice and builds in the build farm

MattSYoung, 2021-06-23:
Correct me if I'm wrong, but in it's current state the `grid_map_pcl` module fails to compile because of its dependency on pcl_ros, which has yet to be ported to ROS2. I've looked through the source code, and unless I missed it I don't think this module acutally uses any code from pcl_ros, it just needs PCL.

I've made the necessary changes in my fork: https://github.com/MattSYoung/grid_map/tree/foxy-devel 

If this is all `grid_map_pcl` needs to be ROS2-ready, can I issue a pull request? And if so, should it be to the `ros2` and/or `foxy-devel` branches?

SteveMacenski, 2021-06-23:
I think that's been covered by the discussion above, but the changes you mention in your PR would be OK by me (make sure to update the rosdep in package.xml). Please remove the specification on the PCL version though. I don't want to have to keep updating that over time and it is explicitly called out in REP2000 what PCL version ships with ROS.

MattSYoung, 2021-06-23:
Fair enough, I've made the requested changes. Now my question is, if I issue a PR to `foxy-devel` should I also issue one to `ros2`? (or any other ROS2 branches for that matter?)

SteveMacenski, 2021-06-23:
Both foxy and ros2 would be good, I believe that is al

roncapat, 2021-09-06:
Any news on ROS2 port status?

mitsudome-r, 2022-01-04:
@SteveMacenski @Marwan99 @maximilianwulf Any updates? Please let me know if there is anything that I can help with.

SteveMacenski, 2022-01-04:
I believe the octomap related work is all that is remaining

wxmerkt, 2022-01-04:
> I believe the octomap related work is all that is remaining

There have been updates to Octomap over the holidays thanks to work by @wep21 - `octomap_mapping` is now ported and awaiting binary release, and `octomap_ros` and `octomap_rviz_plugins` have been released into ROS2. Is there something else missing from the octomap-side of things?

SteveMacenski, 2022-01-04:
Essentially just enabling the things that use them here, if anything.

wep21, 2022-01-06:
@SteveMacenski @wxmerkt #333 enables octomap demo.

SteveMacenski, 2022-01-12:
All items are complete, closing ticket!

closed 2022-01-12 by SteveMacenski

wep21, 2022-01-12:
@SteveMacenski Do you have any plan to release binary for `galactic`?

fmrico, 2025-04-16:
Hi @SteveMacenski, Any plans for releasing binaries for `Humble` or `Rolling`?