ANYbotics/grid_map #287: Problem with grid_map installation in melodic

nPellejero, 2021-02-09:
I am compiling my ros package which depends on grid_map in a NVidia Jetson TX2 with ubuntu 18.04. grid_map is installed from ubuntu repositories along the others ROS dependencies. OpenCV is installed in /usr/local , so includes are in /usr/local/include/opencv and not in /usr/include/opencv  

When compiling, catking throws the following error:

> CMake Error at /opt/ros/melodic/share/grid_map_cv/cmake/grid_map_cvConfig.cmake:113 (message):
  Project 'grid_map_cv' specifies '/usr/include/opencv' as an include dir,
  which is not found.  It does neither exist as an absolute directory nor in
  '${{prefix}}//usr/include/opencv'.  Check the issue tracker
  'http://github.com/anybotics/grid_map/issues' and consider creating a
  ticket if the problem has not been reported yet.

How can I indicate to grid_map to look in /usr/local/include ? is there any way ? I am missing something ?  If possible, I would like to avoid to compile it from source.

Thanks,
Nicolas.

maximilianwulf, 2021-02-09:
Hey @nPellejero, you can add the path to the `find_packag` macro here: https://github.com/ANYbotics/grid_map/blob/master/grid_map_cv/CMakeLists.txt#L15. 

Check out this stackoverflow question: https://stackoverflow.com/questions/49816206/cmake-find-package-specify-path

nPellejero, 2021-02-17:
Ok, then I assume that it is not possible to do it without compiling from source.

Thanks,

closed 2021-02-17 by nPellejero