ANYbotics/grid_map #288: Missing triangles (directional behavior in rviz)

pvazteixeira, 2021-03-02:
I'm building an elevation map using data from an altimeter and I have noticed some interesting behavior in `rviz` - it appears that the triangles in the resulting grid map are dependent on the orientation of a segment. Here is an example, showing both the grid map and the corresponding grid cells (to highlight coverage):

[image]

While on the upper right there are segments of a 'strip', on the upper left there are only half of the triangles. You can see the same behavior on the lower half of the trajectory. In short,▗▛  will result in only one triangle, but ▜▖   will correctly result in two (corner case in [`computeVisualization`](https://github.com/ANYbotics/grid_map/blob/539f9db782e73cd48a112a6a72f3dbc11868d80f/grid_map_rviz_plugin/src/GridMapVisual.cpp#L51),139-215?).

maximilianwulf, 2021-03-07:
Hey @pvazteixeira I think you see the effect of this condition: https://github.com/ANYbotics/grid_map/blob/539f9db782e73cd48a112a6a72f3dbc11868d80f/grid_map_rviz_plugin/src/GridMapVisual.cpp#L180

Do you want different behavior?

pvazteixeira, 2021-03-16:
Hey @maximilianwulf  - thanks for the follow-up and sorry for the delay!

Yes, I think that segment of the code is what is behind this. Maybe I am missing something, but in the example above there is sufficient data to draw two triangles instead of just one (in the top left part of the image), no? If so, then `vertexIndices` is one element short once it gets to 180, so the issue would be in 166:176.

maximilianwulf, 2021-05-31:
Sorry for the delay from my side. That could very well be, do you have a good example in mind to test this, i.e. share the data?