ANYbotics/grid_map #289: grid_map_rviz_plugin: Include files installed to wrong destination

pgorczak, 2021-03-18:
When trying to include the modified MessageFilterDisplay class from my own plugin, I noticed that the rviz plugin's headers are installed to *include/grid_map_rviz_plugin/grid_map_rviz_plugin* instead of *include/grid_map_rviz_plugin*. This breaks the include statement in [*frame_manager.h*](https://github.com/ANYbotics/grid_map/blob/master/grid_map_rviz_plugin/include/grid_map_rviz_plugin/modified/message_filter_display.h#L37).

Could this be caused by the lack of a trailing slash in the CMakeFile's [install command](https://github.com/ANYbotics/grid_map/blob/master/grid_map_rviz_plugin/CMakeLists.txt#L92)?

maximilianwulf, 2021-05-31:
Thank you for the issue. The CMakeFiles install command follows our development guidelines. Let me follow up on this.

maximilianwulf, 2021-05-31:
Doesn't our cmake follow the official recommendation? http://wiki.ros.org/catkin/CMakeLists.txt see `Installing header files`.

maximilianwulf, 2021-05-31:
Yes, that is very much correct. I will fix it. Thank you for the issue submission.