ANYbotics/grid_map #290: Get top left and bottom right corner from grid map

lgulich, 2021-03-24:
Hi 
First of all thank you for open-sourcing this great tool! 

I am trying to extract the top left and bottom right corner of a grid map. I tried doing this with
```c++
const auto top_left = grid_map.getPosition() + grid_map.getLength() / 2.;
const auto bottom_left = grid_map.getPosition() - grid_map.getLength() / 2.;
```
but unfortunately this does not work when the map is `moved`. How would one one extract these points such that it is also correct for moved maps?

Many thanks in advance!

lgulich, 2021-03-26:
Found the solution. As reference for a future reader:
```c++
const Index top_left_idx = grid_map.getStartIndex();
Position top_left;
grid_map.getPosition(top_left_idx, top_left);
const Position bottom_right = top_left - grid_map.getLength().matrix();
```

closed 2021-03-26 by lgulich

maximilianwulf, 2021-05-31:
Thank you for the submission, you could consider also adding it to the README.