ANYbotics/grid_map #292: problem while installing grid_map_cv package:     fatal error: filters/filter_base.hpp: can't find that file or directory

mark9714, 2021-04-07:
[image]

When I try to install grid_map,I've got some problem like this.Did any body have something like this?Actually,I don't know what the filters library is.

bach05, 2021-04-08:
Hi, 
I had the some problem. I solved by substitute in every file `<filters/*.hpp>` with `<filters/*.h>`.

It works fine up to now.

mark9714, 2021-04-08:
@bach05 thank you so much!!I have solved the problem with your help.

closed 2021-04-08 by mark9714

fantasyplus, 2021-04-19:
still not help..

maximilianwulf, 2021-05-31:
Just to be sure, you experienced these issues with melodic, right?

rosrookie, 2022-03-28:
> Hi, I had the some problem. I solved by substitute in every file `<filters/*.hpp>` with `<filters/*.h>`.
> 
> It works fine up to now.

Works for me too, thanks. This happens on ros melodic.

WfHit, 2023-10-10:
So, using " _ln -s /opt/ros/melodic/include/filters/filter_base.h /opt/ros/melodic/include/filters/filter_base.hpp_ " seems more convenient ^^