ANYbotics/grid_map #300: catkin_make command  error

MR-Uranyum, 2021-05-16:
I installed everything as installation say, but I tried building it from source with catkin_make run_tests_grid_map_core run_tests_grid_map_ros command, I have got file not found error. so what am I doing wrong, I left a picture to see my error ,
thank you for your support,
with kind regards.
  
[image]

rosyapril, 2021-05-18:
same error here

wxmerkt, 2021-05-18:
The header is named `filter_base.hpp` on Noetic and `filter_base.h` on Melodic. If you are on an older distribution, you may have to adjust it correspondingly

rosyapril, 2021-05-18:
> The header is named `filter_base.hpp` on Noetic and `filter_base.h` on Melodic. If you are on an older distribution, you may have to adjust it correspondingly

it works! Thanks!

MR-Uranyum, 2021-05-22:
thank you so much, it solved my problem

maximilianwulf, 2021-05-31:
The fix will be merged into the `melodic-devel` branch, see here https://github.com/ANYbotics/grid_map/pull/298