ANYbotics/grid_map #302: How to get "grid_map_costmap_2d" in ROS Kinetic?

maity-kgz, 2021-05-28:
Hi, I am using grid_map in ROS Kinetic. I have installed it from Deb repository. But the grid_map_costmap_2d package doesn't come installed with the grid_map. 
Building from the source in the Kinetic env throws a lot of compiling errors. Due to the compatibility issues with TF and etc. Also, there is no Kinetic branch in the repo.
Is there any chance/way to get grid_map_costmap_2d in the ROS Kinetic?
Many thanx.

maity-kgz, 2021-05-28:
I was able to use "grid_map_costmap_2d" under the Kinetic env after the following steps.
1. Install core grid_map packages from the Deb repo. 
2. Copy to the local catkin_ws only the "grid_map_costmap_2d" folder. 
3. Build the "grid_map_costmap_2d"  with the "catkin_make" cmd. Builds without errors.
4. source local catkin_ws devel space 
5. include "grid_map_costmap_2d" in the find_package (CMakeLists.txt).

maximilianwulf, 2021-05-31:
Thank you for the instructions. We currently don't maintain kinetic.

closed 2021-05-31 by maximilianwulf

congphase, 2021-07-13:
@maity-kgz Instead of installing from source, you can install it with pre-built binary by 

` sudo apt install ros-kinetic-costmap-2d `

Have you tried it?