ANYbotics/grid_map #325: pointcloud from rtabmap works but not from gazebo

The-Developer-02, 2021-11-03:
Hi, I'm trying to create a map from the point cloud produced by gazebo, if I run it through rtabmap first i end up with a grid map but if I take it directly from Gazebo it fails, without telling me why. Both systems output a PointCloud2, so I don't know why one works and not the other. Thanks for your help, error pasted below:

`[ INFO] [1635901150.006727255, 34.707000000]: Initialized map geometry

[ INFO] [1635901150.006807217, 34.707000000]: Initialization took: 0.524 sec

[ INFO] [1635901150.009073329, 34.707000000]: Started adding layer: elevation

[grid_map_pcl_loader_node-1] process has died [pid 130835, exit code -11, cmd /home/sam/catkin_ws/devel/lib/grid_map_pcl/grid_map_pcl_loader_node __name:=grid_map_pcl_loader_node __log:=/home/sam/.ros/log/627cdb86-3c3f-11ec-948f-0599ff4a7caf/grid_map_pcl_loader_node-1.log].

log file: /home/sam/.ros/log/627cdb86-3c3f-11ec-948f-0599ff4a7caf/grid_map_pcl_loader_node-1*.log

all processes on machine have died, roslaunch will exit
`

maximilianwulf, 2021-11-25:
Can you build it in `DEBUG` to give more insights? 

Could you compare the fields both point clouds have?

The-Developer-02, 2021-12-04:
Thanks for getting back to me, I'm still pretty new to this, I think this is what you wanted, I've copied as much as my console would allow, I had to attach as a file as it was too many characters for github

How would you like me to compare them, below is the information I get on each just before the process point cloud command is called, the top info is from rtabmap the bottom is from gazebo, I've tried adjusting the height and number of points, but they don't work and I'm not sure how to adjust any of the other variables.

`cloud to processheader: seq: 0 stamp: 77141000 frame_id: map

points[]: 4708
width: 4708
height: 1
is_dense: 1
sensor origin (xyz): [0, 0, 0] / orientation (xyzw): [0, 0, 0, 1]

cloud to processheader: seq: 12957 stamp: 647944000 frame_id: camera_link

points[]: 10000
width: 100
[built with debug.txt](https://github.com/ANYbotics/grid_map/files/7653991/built.with.debug.txt)
height: 100
is_dense: 0
sensor origin (xyz): [0, 0, 0] / orientation (xyzw): [0, 0, 0, 1]
[built with debug.txt](https://github.com/ANYbotics/grid_map/files/7653992/built.with.debug.txt)
`

maximilianwulf, 2021-12-06:
Hey, sorry, I meant to build it in debug, and then let it crash again `[grid_map_pcl_loader_node-1] process has died `. I think your logs are only related to the build itself.

The-Developer-02, 2021-12-06:
I Built it in debug and then ran it again and I get the same message, no more information at all

`[ INFO] [1638802974.316650710, 209.135000000]: Initialized map geometry
[ INFO] [1638802974.316701183, 209.135000000]: Initialization took: 0.583 sec
[ INFO] [1638802974.317968683, 209.136000000]: Started adding layer: elevation
[grid_map_pcl_loader_node-1] process has died [pid 82330, exit code -11, cmd /home/sam/catkin_ws/devel/lib/grid_map_pcl/grid_map_pcl_loader_node __name:=grid_map_pcl_loader_node __log:=/home/sam/.ros/log/01ad134a-56a5-11ec-9cf0-9d5b6d1516cc/grid_map_pcl_loader_node-1.log].
log file: /home/sam/.ros/log/01ad134a-56a5-11ec-9cf0-9d5b6d1516cc/grid_map_pcl_loader_node-1*.log
all processes on machine have died, roslaunch will exit
`

maximilianwulf, 2021-12-06:
Can you try to open it with a debugger, such as `gdb` ?

The-Developer-02, 2021-12-06:
I think I ran it correctly, this is what I got

`[New Thread 0x7fffe4528700 (LWP 127205)]
[New Thread 0x7fffe3d27700 (LWP 127206)]
[New Thread 0x7fffe3526700 (LWP 127207)]
[New Thread 0x7fffe2d25700 (LWP 127208)]
[ INFO] [1638826746.395586034]: hello from ROS Node: /grid_map_pcl_loader_node
[ INFO] [1638826746.848448682, 23902.467000000]: cloud to processheader: seq: 1475 stamp: 23902233000 frame_id: map

points[]: 640000
width: 800
height: 800
is_dense: 0
sensor origin (xyz): [0, 0, 0] / orientation (xyzw): [0, 0, 0, 1]

[ INFO] [1638826746.848553635, 23902.467000000]: Preprocessing of the pointcloud started
[ INFO] [1638826746.923309713, 23902.531000000]: Preprocessing and filtering finished
[ INFO] [1638826747.170858662, 23902.752000000]: Grid map dimensions: 5.5 x 3.3
[ INFO] [1638826747.171413102, 23902.752000000]: Grid map resolution: 0.1
[ INFO] [1638826747.171685549, 23902.752000000]: Grid map num cells: 55 x 33
[ INFO] [1638826747.171921164, 23902.754000000]: Initialized map geometry
[ INFO] [1638826747.172342390, 23902.754000000]: Initialization took: 0.323 sec
[ INFO] [1638826747.173796256, 23902.756000000]: Started adding layer: elevation
--Type <RET> for more, q to quit, c to continue without paging--c

Thread 1 "grid_map_pcl_lo" received signal SIGSEGV, Segmentation fault.
0x00007ffff7f54eb8 in std::vector<boost::shared_ptr<pcl::PointCloud<pcl::PointXYZ> >, std::allocator<boost::shared_ptr<pcl::PointCloud<pcl::PointXYZ> > > >::operator[] (this=0x554958c20be0, __n=18446744071562067968) at /usr/include/c++/9/bits/stl_vector.h:1043
1043		return *(this->_M_impl._M_start + __n);
`

maximilianwulf, 2021-12-07:
Hm I think the software crashes while accessing the point cloud with an invalid index, can you display the point cloud coming out of rtabmap in rviz to verify the point cloud exists?

The-Developer-02, 2021-12-07:
Yes, both display absolutely fine in rviz

maximilianwulf, 2021-12-08:
Hm, could you share a small bag with the point cloud.  I am a bit out of ideas here what could be the breaking issue.

The-Developer-02, 2021-12-08:
here you go, I think that has worked, I've done both the rtabmap point cloud (which works) and the gazebo point cloud (which doesn't), they're not very long unfortunately as the file would have been too large to upload

[gazebo.bag.gz](https://github.com/ANYbotics/grid_map/files/7675702/gazebo.bag.gz)
[rtabmap.bag.gz](https://github.com/ANYbotics/grid_map/files/7675704/rtabmap.bag.gz)

The-Developer-02, 2022-02-01:
Have you managed to find anything out? @maximilianwulf

maximilianwulf, 2022-02-01:
Hey @The-Developer-02 , no yet sorry.