ANYbotics/grid_map #328: grid_map_ros

kennydedavion, 2021-11-23:
Good day. I have a big problem with your package. When I first wanted to do catkin_make, catkin announced dual grid_map_ros packages. Because I downloaded it. So I deleted a package that is not part of your package. But now catkin is unable to find the dependency of grid_map_cv.
How the hell is that possible? When it's all in one package?  Apropo, Your "sudo apt-get install ros-melodic-grid-map" cannot be installed on raspbian using this display, please edit. THX J

maximilianwulf, 2021-11-25:
Hi, can you try this fix? https://github.com/ANYbotics/grid_map/pull/327

kennydedavion, 2021-11-25:
Thanks, but it didn't help.
 This package is already downloaded. 
More precisely, I screened your entire "grid_map" repo. 

In the folder: grid_map_ros, I created the build folder.
 I ran the "cmake .." display inside this folder and the installer can't find the "grid_map_cv" DIR, specifically grid_map_cvConfig.cmake.
 So I can see in the "grid_map_cv" folder in the SRC folder the procedure : rosdep install -y --from-paths. --ignore-src - melodic table.
 In the "grid_map_cv" folder, I created the "build" folder, in which I ran the "cmake .." display. Successful. then run "make", 100 percent build.
 Then run "sudo make install", 100 percent build and you see it there too? 
The compilation of the "grid_map_ros" package still reports an error after all. 

`Install the project...
-- Install configuration: ""
-- Installing: /usr/local/_setup_util.py
-- Up-to-date: /usr/local/env.sh
-- Up-to-date: /usr/local/setup.bash
-- Up-to-date: /usr/local/local_setup.bash
-- Up-to-date: /usr/local/setup.sh
-- Up-to-date: /usr/local/local_setup.sh
-- Up-to-date: /usr/local/setup.zsh
-- Up-to-date: /usr/local/local_setup.zsh
-- Up-to-date: /usr/local/.rosinstall
-- Installing: /usr/local/lib/pkgconfig/grid_map_cv.pc
-- Up-to-date: /usr/local/share/grid_map_cv/cmake/grid_map_cvConfig.cmake
-- Up-to-date: /usr/local/share/grid_map_cv/cmake/grid_map_cvConfig-version.cmake
-- Up-to-date: /usr/local/share/grid_map_cv/package.xml
-- Up-to-date: /usr/local/lib/libgrid_map_cv.so
-- Up-to-date: /usr/local/include/grid_map_cv
-- Up-to-date: /usr/local/include/grid_map_cv/InpaintFilter.hpp
-- Up-to-date: /usr/local/include/grid_map_cv/GridMapCvProcessing.hpp
-- Up-to-date: /usr/local/include/grid_map_cv/grid_map_cv.hpp
-- Up-to-date: /usr/local/include/grid_map_cv/GridMapCvConverter.hpp
-- Up-to-date: /usr/local/share/grid_map_cv/filter_plugins.xml

I'm helpless. thank you for any help. J
`

maximilianwulf, 2021-11-26:
Can you try to build it with `catkin build` instead of `catkin_make`?

kennydedavion, 2021-11-28:
Im try catkin_build :