ANYbotics/grid_map #336: Integrate a static gridmap into a local gridmap

BIRL-xu, 2022-01-27:
Hi, I have a static gridmap with size 100mx100m converted from ros occupancy grid map, and a local gridmap with size 20mx10m built online. The local gridmap would be moved with robot.
I want to integrate the grid value in the overlapping area between two maps into the local gridmap. Could you give me some tips how to implement this? I'm a beginner for this great library. Thank you very much!

maximilianwulf, 2022-01-27:
Hi @BIRL-xu, 

Interesting question, I think you have to work out these questions: 

- Align the maps first, is this given? 
- Get the corresponding values of the overlapping grid maps. You should be able to calculate overlapping grid cells with `.position()` function and such which can be found in the core library. 
- Once you have the overlapping values, you have to think about how to fuse the values. The simplest solution is just averaging the height values. But you can also do weighted average, kalman filtering, etc.

I hope this helps.

BIRL-xu, 2022-01-28:
Thank you for your reply, @maximilianwulf 

- I know the transformation between two maps, so I can transform the point in static map to the one in local map.
- I'm not sure if there is an efficient way to find out the overlapping area between two maps, I'm learning class `PolygonIterator` and `SubmapIterator` for this. 
- Actually, I'm doing a cost map just like Costmap2D, so I think fuse the grids with the maximum cost is ok.