ANYbotics/grid_map #345: Compilation issue on foxy

Vinuu428, 2022-03-22:
home/user/gridmap_dep/src/grid_map/grid_map_ros/src/GridMapRosConverter.cpp:278:27: error: ‘to_yaml’ is not a member of ‘geometry_msgs::msg’
  278 |       geometry_msgs::msg::to_yaml(occupancyGrid.info.origin.orientation).c_str());
      |                           ^~~~~~~
/home/user/gridmap_dep/src/grid_map/grid_map_ros/src/GridMapRosConverter.cpp: In static member function ‘static bool grid_map::GridMapRosConverter::fromCostmap(const Costmap&, const string&, grid_map::GridMap&)’:
/home/user/gridmap_dep/src/grid_map/grid_map_ros/src/GridMapRosConverter.cpp:377:27: error: ‘to_yaml’ is not a member of ‘geometry_msgs::msg’
  377 |       geometry_msgs::msg::to_yaml(costmap.metadata.origin.orientation).c_str());
      |                           ^~~~~~~
/home/user/gridmap_dep/src/grid_map/grid_map_ros/src/GridMapRosConverter.cpp: In static member function ‘static bool grid_map::GridMapRosConverter::saveToBag(const grid_map::GridMap&, const string&, const string&)’:
/home/user/gridmap_dep/src/grid_map/grid_map_ros/src/GridMapRosConverter.cpp:672:20: error: ‘StorageOptions’ is not a member of ‘rosbag2_storage’
  672 |   rosbag2_storage::StorageOptions storage_options;
      |                    ^~~~~~~~~~~~~~
/home/user/gridmap_dep/src/grid_map/grid_map_ros/src/GridMapRosConverter.cpp:672:20: note: suggested alternatives:
In file included from /opt/ros/foxy/include/rosbag2_cpp/writer.hpp:25,
                 from /home/user/gridmap_dep/src/grid_map/grid_map_ros/src/GridMapRosConverter.cpp:13:
/opt/ros/foxy/include/rosbag2_cpp/storage_options.hpp:23:8: note:   ‘rosbag2_cpp::StorageOptions’
   23 | struct StorageOptions
      |        ^~~~~~~~~~~~~~
In file included from /usr/include/eigen3/Eigen/Core:363,
                 from /home/user/gridmap_dep/install/grid_map_core/include/grid_map_core/TypeDefs.hpp:13,
                 from /home/user/gridmap_dep/install/grid_map_core/include/grid_map_core/grid_map_core.hpp:12,
                 from /home/user/gridmap_dep/install/grid_map_cv/include/grid_map_cv/grid_map_cv.hpp:12,
                 from /home/user/gridmap_dep/src/grid_map/grid_map_ros/src/GridMapRosConverter.cpp:21:
/usr/include/eigen3/Eigen/src/Core/util/Constants.h:318:6: note:   ‘Eigen::StorageOptions’
  318 | enum StorageOptions {
      |      ^~~~~~~~~~~~~~
/home/user/gridmap_dep/src/grid_map/grid_map_ros/src/GridMapRosConverter.cpp:673:3: error: ‘storage_options’ was not declared in this scope
  673 |   storage_options.uri = pathToBag;
      |   ^~~~~~~~~~~~~~~
/home/user/gridmap_dep/src/grid_map/grid_map_ros/src/GridMapRosConverter.cpp: In static member function ‘static bool grid_map::GridMapRosConverter::loadFromBag(const string&, const string&, grid_map::GridMap&)’:
/home/user/gridmap_dep/src/grid_map/grid_map_ros/src/GridMapRosConverter.cpp:708:20: error: ‘StorageOptions’ is not a member of ‘rosbag2_storage’
  708 |   rosbag2_storage::StorageOptions storage_options;
      |                    ^~~~~~~~~~~~~~
/home/user/gridmap_dep/src/grid_map/grid_map_ros/src/GridMapRosConverter.cpp:708:20: note: suggested alternatives:
In file included from /opt/ros/foxy/include/rosbag2_cpp/writer.hpp:25,
                 from /home/user/gridmap_dep/src/grid_map/grid_map_ros/src/GridMapRosConverter.cpp:13:
/opt/ros/foxy/include/rosbag2_cpp/storage_options.hpp:23:8: note:   ‘rosbag2_cpp::StorageOptions’
   23 | struct StorageOptions
      |        ^~~~~~~~~~~~~~
In file included from /usr/include/eigen3/Eigen/Core:363,
                 from /home/user/gridmap_dep/install/grid_map_core/include/grid_map_core/TypeDefs.hpp:13,
                 from /home/user/gridmap_dep/install/grid_map_core/include/grid_map_core/grid_map_core.hpp:12,
                 from /home/user/gridmap_dep/install/grid_map_cv/include/grid_map_cv/grid_map_cv.hpp:12,
                 from /home/user/gridmap_dep/src/grid_map/grid_map_ros/src/GridMapRosConverter.cpp:21:
/usr/include/eigen3/Eigen/src/Core/util/Constants.h:318:6: note:   ‘Eigen::StorageOptions’
  318 | enum StorageOptions {
      |      ^~~~~~~~~~~~~~
/home/user/gridmap_dep/src/grid_map/grid_map_ros/src/GridMapRosConverter.cpp:709:3: error: ‘storage_options’ was not declared in this scope
  709 |   storage_options.uri = pathToBag;
      |   ^~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/grid_map_ros.dir/build.make:63: CMakeFiles/grid_map_ros.dir/src/GridMapRosConverter.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:130: CMakeFiles/grid_map_ros.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< grid_map_ros [2.77s, exited with code 2]
Aborted  <<< grid_map_costmap_2d [3.83s]                                

Summary: 6 packages finished [4.26s]
  1 package failed: grid_map_ros
  1 package aborted: grid_map_costmap_2d
  2 packages had stderr output: grid_map_costmap_2d grid_map_ros
  7 packages not processed

wwdafandian, 2022-04-06:
Hi, @Vinuu428  I got the exact same error. Have you figured it out?

Vinuu428, 2022-04-07:
Yes. my ROS2 version and git branch are not compatible. choose the right branch and pull the code

closed 2022-04-07 by Vinuu428