ANYbotics/grid_map #355: Cmake error about TBB

Krasjet-Yu, 2022-06-14:
Why this error happens after the installation ?

`CMake Error at grid_map/grid_map_filters/CMakeLists.txt:35 (find_package):
  By not providing "FindTBB.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "TBB", but
  CMake did not find one.

  Could not find a package configuration file provided by "TBB" (requested
  version 2020.1) with any of the following names:

    TBBConfig.cmake
    tbb-config.cmake

  Add the installation prefix of "TBB" to CMAKE_PREFIX_PATH or set "TBB_DIR"
  to a directory containing one of the above files.  If "TBB" provides a
  separate development package or SDK, be sure it has been installed.
`

maximilianwulf, 2022-06-15:
Hey, Intel oneAPI changed its API of TBB. So we decided to fix it for the officially supported Ubuntu 20.04 debian package, and for noetic. 

On what system are you compiling?

maximilianwulf, 2022-06-15:
Basically, `sudo apt install libtbb-dev` could/should solve the issue. 

We only support Ubuntu 20.04 and Noetic at the moment. 

On my machine I have the following installed: 

```
[~] apt show libtbb-dev   
Package: libtbb-dev
Version: 2020.1-2
Priority: extra
Section: universe/libdevel
Source: tbb
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Debian Science Maintainers <debian-science-maintainers@lists.alioth.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 2,019 kB
Depends: libtbb2 (= 2020.1-2)
Suggests: libtbb-doc
Homepage: https://www.threadingbuildingblocks.org/
Download-Size: 273 kB
APT-Manual-Installed: yes
APT-Sources: http://ch.archive.ubuntu.com/ubuntu focal/universe amd64 Packages
Description: parallelism library for C++ - development files
 TBB is a library that helps you leverage multi-core processor
 performance without having to be a threading expert. It represents a
 higher-level, task-based parallelism that abstracts platform details
 and threading mechanism for performance and scalability.
 .
 (Note: if you are a user of the i386 architecture, i.e., 32-bit Intel
 or compatible hardware, this package only supports Pentium4-compatible
 and higher processors.)
 .
 This package includes the TBB development files.
 
```

Krasjet-Yu, 2022-06-15:
Finally I commented out the TBB.

maximilianwulf, 2022-06-17:
Okay, also valid. 

I guess 

```
find_package(TBB REQUIRED)
```

would have worked in your case. But it is not necessary.

closed 2022-06-17 by maximilianwulf

giulioturrisi, 2024-04-04:
Hi, 

i'm using grid_map for ubuntu 22, ros noetic. The latest fix in commit 8a95e56 resolves the problem for compilation with TBB. The package seems to run together with elevation_mapping.