ANYbotics/grid_map #357: ubuntu 20.04 cmake error

EricChen2013, 2022-06-16:
--- stderr: grid_map_cv                 
/home/ericchen/AutowareAuto/src/external/grid_map/grid_map_cv/src/InpaintFilter.cpp: In instantiation of ‘bool grid_map::InpaintFilter<T>::configure() [with T = grid_map::GridMap]’:
/home/ericchen/AutowareAuto/src/external/grid_map/grid_map_cv/src/InpaintFilter.cpp:37:6:   required from here
/home/ericchen/AutowareAuto/src/external/grid_map/grid_map_cv/src/InpaintFilter.cpp:39:38: error: ‘class grid_map::InpaintFilter<grid_map::GridMap>’ has no member named ‘param_prefix_’
   39 |   ParameterReader param_reader(this->param_prefix_, this->params_interface_);
      |                                ~~~~~~^~~~~~~~~~~~~
/home/ericchen/AutowareAuto/src/external/grid_map/grid_map_cv/src/InpaintFilter.cpp:39:59: error: ‘class grid_map::InpaintFilter<grid_map::GridMap>’ has no member named ‘params_interface_’
   39 |   ParameterReader param_reader(this->param_prefix_, this->params_interface_);
      |                                                     ~~~~~~^~~~~~~~~~~~~~~~~
In file included from /opt/ros/foxy/include/rclcpp/service.hpp:34,
                 from /opt/ros/foxy/include/rclcpp/callback_group.hpp:25,
                 from /opt/ros/foxy/include/rclcpp/any_executable.hpp:20,
                 from /opt/ros/foxy/include/rclcpp/memory_strategy.hpp:24,
                 from /opt/ros/foxy/include/rclcpp/memory_strategies.hpp:18,
                 from /opt/ros/foxy/include/rclcpp/executor_options.hpp:20,
                 from /opt/ros/foxy/include/rclcpp/executor.hpp:33,
                 from /opt/ros/foxy/include/rclcpp/executors/multi_threaded_executor.hpp:26,
                 from /opt/ros/foxy/include/rclcpp/executors.hpp:21,
                 from /opt/ros/foxy/include/rclcpp/rclcpp.hpp:146,
                 from /home/ericchen/AutowareAuto/src/external/grid_map/grid_map_cv/src/InpaintFilter.cpp:14:
/home/ericchen/AutowareAuto/src/external/grid_map/grid_map_cv/src/InpaintFilter.cpp:43:13: error: ‘class grid_map::InpaintFilter<grid_map::GridMap>’ has no member named ‘logging_interface_’
   43 |       this->logging_interface_->get_logger(), "InpaintRadius filter did not find param radius.");
      |       ~~~~~~^~~~~~~~~~~~~~~~~~
In file included from /opt/ros/foxy/include/rclcpp/client.hpp:40,
                 from /opt/ros/foxy/include/rclcpp/callback_group.hpp:23,
                 from /opt/ros/foxy/include/rclcpp/any_executable.hpp:20,
                 from /opt/ros/foxy/include/rclcpp/memory_strategy.hpp:24,
                 from /opt/ros/foxy/include/rclcpp/memory_strategies.hpp:18,
                 from /opt/ros/foxy/include/rclcpp/executor_options.hpp:20,
                 from /opt/ros/foxy/include/rclcpp/executor.hpp:33,
                 from /opt/ros/foxy/include/rclcpp/executors/multi_threaded_executor.hpp:26,
                 from /opt/ros/foxy/include/rclcpp/executors.hpp:21,
                 from /opt/ros/foxy/include/rclcpp/rclcpp.hpp:146,
                 from /home/ericchen/AutowareAuto/src/external/grid_map/grid_map_cv/src/InpaintFilter.cpp:14:
/home/ericchen/AutowareAuto/src/external/grid_map/grid_map_cv/src/InpaintFilter.cpp:43:13: error: ‘class grid_map::InpaintFilter<grid_map::GridMap>’ has no member named ‘logging_interface_’
   43 |       this->logging_interface_->get_logger(), "InpaintRadius filter did not find param radius.");
      |       ~~~~~~^~~~~~~~~~~~~~~~~~
/home/ericchen/AutowareAuto/src/external/grid_map/grid_map_cv/src/InpaintFilter.cpp:43:13: error: ‘class grid_map::InpaintFilter<grid_map::GridMap>’ has no member named ‘logging_interface_’
   43 |       this->logging_interface_->get_logger(), "InpaintRadius filter did not find param radius.");
      |       ~~~~~~^~~~~~~~~~~~~~~~~~
In file included from /opt/ros/foxy/include/rclcpp/service.hpp:34,
                 from /opt/ros/foxy/include/rclcpp/callback_group.hpp:25,
                 from /opt/ros/foxy/include/rclcpp/any_executable.hpp:20,
                 from /opt/ros/foxy/include/rclcpp/memory_strategy.hpp:24,
                 from /opt/ros/foxy/include/rclcpp/memory_strategies.hpp:18,
                 from /opt/ros/foxy/include/rclcpp/executor_options.hpp:20,
                 from /opt/ros/foxy/include/rclcpp/executor.hpp:33,
                 from /opt/ros/foxy/include/rclcpp/executors/multi_threaded_executor.hpp:26,
                 from /opt/ros/foxy/include/rclcpp/executors.hpp:21,
                 from /opt/ros/foxy/include/rclcpp/rclcpp.hpp:146,
                 from /home/ericchen/AutowareAuto/src/external/grid_map/grid_map_cv/src/InpaintFilter.cpp:14:
/home/ericchen/AutowareAuto/src/external/grid_map/grid_map_cv/src/InpaintFilter.cpp:43:13: error: ‘class grid_map::InpaintFilter<grid_map::GridMap>’ has no member named ‘logging_interface_’
   43 |       this->logging_interface_->get_logger(), "InpaintRadius filter did not find param radius.");
      |       ~~~~~~^~~~~~~~~~~~~~~~~~
In file included from /opt/ros/foxy/include/rclcpp/client.hpp:40,
                 from /opt/ros/foxy/include/rclcpp/callback_group.hpp:23,
                 from /opt/ros/foxy/include/rclcpp/any_executable.hpp:20,
                 from /opt/ros/foxy/include/rclcpp/memory_strategy.hpp:24,
                 from /opt/ros/foxy/include/rclcpp/memory_strategies.hpp:18,
                 from /opt/ros/foxy/include/rclcpp/executor_options.hpp:20,
                 from /opt/ros/foxy/include/rclcpp/executor.hpp:33,
                 from /opt/ros/foxy/include/rclcpp/executors/multi_threaded_executor.hpp:26,
                 from /opt/ros/foxy/include/rclcpp/executors.hpp:21,
                 from /opt/ros/foxy/include/rclcpp/rclcpp.hpp:146,
                 from /home/ericchen/AutowareAuto/src/external/grid_map/grid_map_cv/src/InpaintFilter.cpp:14:
/home/ericchen/AutowareAuto/src/external/grid_map/grid_map_cv/src/InpaintFilter.cpp:48:24: error: ‘class grid_map::InpaintFilter<grid_map::GridMap>’ has no member named ‘logging_interface_’
   48 |     RCLCPP_ERROR(this->logging_interface_->get_logger(), "Radius must be greater than zero.");
      |                  ~~~~~~^~~~~~~~~~~~~~~~~~
/home/ericchen/AutowareAuto/src/external/grid_map/grid_map_cv/src/InpaintFilter.cpp:48:24: error: ‘class grid_map::InpaintFilter<grid_map::GridMap>’ has no member named ‘logging_interface_’
   48 |     RCLCPP_ERROR(this->logging_interface_->get_logger(), "Radius must be greater than zero.");
[... 134 lines cut ...]
                 from /opt/ros/foxy/include/rclcpp/rclcpp.hpp:146,
                 from /home/ericchen/AutowareAuto/src/external/grid_map/grid_map_cv/src/InpaintFilter.cpp:14:
/home/ericchen/AutowareAuto/src/external/grid_map/grid_map_cv/src/InpaintFilter.cpp:43:13: error: ‘class grid_map::InpaintFilter<grid_map::GridMap>’ has no member named ‘logging_interface_’
   43 |       this->logging_interface_->get_logger(), "InpaintRadius filter did not find param radius.");
      |       ~~~~~~^~~~~~~~~~~~~~~~~~
In file included from /opt/ros/foxy/include/rclcpp/client.hpp:40,
                 from /opt/ros/foxy/include/rclcpp/callback_group.hpp:23,
                 from /opt/ros/foxy/include/rclcpp/any_executable.hpp:20,
                 from /opt/ros/foxy/include/rclcpp/memory_strategy.hpp:24,
                 from /opt/ros/foxy/include/rclcpp/memory_strategies.hpp:18,
                 from /opt/ros/foxy/include/rclcpp/executor_options.hpp:20,
                 from /opt/ros/foxy/include/rclcpp/executor.hpp:33,
                 from /opt/ros/foxy/include/rclcpp/executors/multi_threaded_executor.hpp:26,
                 from /opt/ros/foxy/include/rclcpp/executors.hpp:21,
                 from /opt/ros/foxy/include/rclcpp/rclcpp.hpp:146,
                 from /home/ericchen/AutowareAuto/src/external/grid_map/grid_map_cv/src/InpaintFilter.cpp:14:
/home/ericchen/AutowareAuto/src/external/grid_map/grid_map_cv/src/InpaintFilter.cpp:72:11: error: ‘class grid_map::InpaintFilter<grid_map::GridMap>’ has no member named ‘logging_interface_’
   72 |     this->logging_interface_->get_logger(), "Inpaint output layer = %s.", outputLayer_.c_str());
      |     ~~~~~~^~~~~~~~~~~~~~~~~~
/home/ericchen/AutowareAuto/src/external/grid_map/grid_map_cv/src/InpaintFilter.cpp:72:11: error: ‘class grid_map::InpaintFilter<grid_map::GridMap>’ has no member named ‘logging_interface_’
   72 |     this->logging_interface_->get_logger(), "Inpaint output layer = %s.", outputLayer_.c_str());
      |     ~~~~~~^~~~~~~~~~~~~~~~~~
make[2]: *** [CMakeFiles/grid_map_cv.dir/build.make:76: CMakeFiles/grid_map_cv.dir/src/InpaintFilter.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:241: CMakeFiles/grid_map_cv.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

maximilianwulf, 2022-06-17:
Can you check if `logging_interface_` exists? 

```
grid_map::InpaintFilter<grid_map::GridMap>’ has no member named ‘logging_interface_’
66 | this->logging_interface_->get_logger(),
```

ArghyaChatterjee, 2022-11-30:
Were you able to solve it ?

heacens, 2023-11-14:
> did anyone solved this problem ？

solved, not installed  filters package 
sudo apt install ros-galactic-filters

Ryanf55, 2024-02-01:
Looks like a missing rosdep, find_package, and link to this package: https://index.ros.org/p/filters/

Ryanf55, 2024-02-12:
On `rolling`, there is a dependency on filters; it builds and links fine, so I'm closing this issue as invalid.

closed 2024-02-12 by Ryanf55