ANYbotics/grid_map #361: Robot-Centric Elevation Mapping + Grid Map SDF

patripfr, 2022-06-27:
Hi,
I'm using the elevation_mapping package to create a grid map from point clouds and I want to use the grid_map_sdf functionality to create an SDF out of it. I'm following the [demo](https://github.com/ANYbotics/grid_map/blob/master/grid_map_demos/src/SdfDemo.cpp#L40-L55) for the conversion and visualization.

While to robot is in the XY origin, this works fine and the elevation map and the sdf are aligned. 
[image]

However, once the robot moves, the location of the sdf does not update correctly and creates wrong values as indicated in the below figure. While the "overlapping" are seems fine, the values in the newly observed area on the left side of the image, are actually created on the right side.
[image]

 Am I missing something here or is the combination of grid_maps from the elevation_mapping packes with the grid_map_sdf package not supported?

maximilianwulf, 2022-06-27:
Looks like a indice mismatch to me. 

How do you trigger re-computation of the sdf after the robot has moved?

patripfr, 2022-06-27:
Calling `convertToDefaultStartIndex()` fixed it! Thanks a lot!

closed 2022-06-27 by patripfr