ANYbotics/grid_map #362: GridMapPclConverter consume really long time from polygonmesh2gridmap

Kin-Zhang, 2022-07-04:
[image]

first, it will build mesh through vdbfusion, but when I convert it to PCL polygon mesh with filtered a number of points also, but it still consumes lots of time, I'm wondering that is it normal for this script converter? through I knew the converter did ray intersection with each mesh on the grid. But it still wried if it has to use so much time...

https://github.com/ANYbotics/grid_map/blob/master/grid_map_pcl/src/GridMapPclConverter.cpp

maximilianwulf, 2022-07-04:
I guess the bottleneck is 

```
  for (const pcl::Vertices& polygon : mesh.polygons) {
```

How big is your mesh?

Kin-Zhang, 2022-07-04:
as the figure shows, the mesh has 1287 vertices, each vertice has three other vertices' id which is the for loop one.

I assume the same thing. but it still costs too much time, just wried... maybe also the ray intersection on all polygons

maximilianwulf, 2022-07-04:
you can try to find the the hotspot with `Intel vtune`.

closed 2022-07-07 by Kin-Zhang