ANYbotics/grid_map #371: Unable to build grid_map because can't find pcl_ros. Under ROS2 Foxy and using the branch ros2

pgarcia-dev, 2022-09-03:
Hello,

I post here the url where I explain the problem I'm facing and I wrote in ROS answers: https://answers.ros.org/question/405833/unable-to-build-grid_map-because-cant-find-pcl_ros/

Thanks

RaamRosik, 2022-09-05:
Use the branch foxy-devel

pgarcia-dev, 2022-09-06:
> Use the branch foxy-devel

Thank you! Now I was able to build grid_map And execute the demos 👌

alexMarFar, 2022-10-28:
Hi! I have the same problem but I cannot seem to solve it. 

I have tried following the Readme from ros2 branch and got the "Package 'pcl_ros' specified with --packages-up-to was not found" message. After finding this issue I tried with th foxy-devel branch, following the read me commands and even modifying the line "wget https://raw.githubusercontent.com/ANYbotics/grid_map/ros2/tools/ros2_dependencies.repos" to "wget https://raw.githubusercontent.com/ANYbotics/grid_map/foxy-devel/tools/ros2_dependencies.repos" in case there was a typo bu still get the "Package 'pcl_ros' specified with --packages-up-to was not found" message. 

Is there something I am doing wrong?

pgarcia-dev, 2022-10-28:
Hello Alejandra,
I finally solved the problem using the branch foxy-devel instead of the ros2 one.

I closed a ticket I opened on Ros answers: https://answers.ros.org/question/405833/unable-to-build-grid_map-because-cant-find-pcl_ros/

Best regards,
Pablo