ANYbotics/grid_map #380: Interpreting GridMap.msg: Convert Elevation Map to Height Map

Quest2GM, 2022-12-19:
Hi @maximilianwulf,

I'd like to know if there's more detailed documentation to this ROS message: https://github.com/anybotics/grid_map/blob/master/grid_map_msgs/msg/GridMap.msg. I can't really understand how the data in here is structured, specifically the grid map data. My end goal is to be able to convert the elevation map to a height map, where the height map is a 2D grayscale image and the elevation at a particular location is encoded by the pixel intensity in the image. I believe that trying to understand this ROS msg is the first step to achieving this.

Any help would be really appreciated.

Thanks,
Quest2GM

Quest2GM, 2022-12-21:
Was able to solve the problem using the elevation_mapping ROS topic directly because it directly gives you height values.

closed 2022-12-21 by Quest2GM

Ion1811, 2024-03-19:
Is there anyway to get the height position of a single cell?

Quest2GM, 2024-03-19:
Why not apply elevation mapping to the entire patch, and then get the height of the specific cell you need?

Ion1811, 2024-03-20:
Yes, sorry, I meant after applying elevation mapping, how to get the value of the elevation coordinate.

Quest2GM, 2024-03-21:
I pulled out some bits and pieces of code that I used a while ago, so I can't really remember the exact details of this, but here's some guidance.

Create rospy subscriber to the elevation mapping topic:
`rospy.Subscriber("/elevation_mapping/elevation_map", GridMap, cb1)`

Callback function for subscriber is defined as such:
```
def cb1(data):
    elev_cb(data.data[0].data, data.inner_start_index, data.outer_start_index)
```
elev_cb is defined as such:
```
def elev_cb(data, s1, s2, ns):

    # Get elevation data (index 0) and roll/flip array according to start indices
    d = np.array(data)   
    d = d.reshape(63,63)
    d = np.roll(d,-s1,axis=0)
    d = np.roll(d,-s2,axis=1)
    d = np.flip(d,axis=1)
    d *= 255
```
From what I can remember, what I did here is basically use the output from `data.data[0].data` in the callback function (which I believe provides some height values) and then using `elev_cb` to convert this to an image heightmap. I had to use np.roll because the inner and outer start index were not defined at the corner of the image so the image needed to be left/right shifted.

I believe for `data.data[0].data`, if you use other indices like 1, 2, 3.. etc instead of the index 0, I believe there's some other data there that related to height values, but I can't remember. You will need to check all the indices and verify that the numbers are reasonable.

Let me know if this works for you.

Ion1811, 2024-04-03:
Thank you so much! I