ANYbotics/grid_map #382: Eigen Plugins not compiled.

MariamH98, 2023-01-13:
I want to use grid_map_core so I built my package using colcon but I keep getting this error;
`fatal error: grid_map_core/eigen_plugins/FunctorsPlugin.hpp: No such file or directory`

I am using the repo rolling on ubuntu 22.04 and my cmake file looks like this:

`cmake_minimum_required(VERSION 3.8)
project(shaft-mapping)

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

# find dependencies
find_package(ament_cmake REQUIRED)
find_package(rclcpp REQUIRED)
find_package(sensor_msgs REQUIRED)
find_package(std_msgs REQUIRED)
find_package(grid_map_core REQUIRED)
find_package(grid_map_cmake_helpers REQUIRED)

if(BUILD_TESTING)
  find_package(ament_lint_auto REQUIRED)
  set(ament_cmake_copyright_FOUND TRUE)
  set(ament_cmake_cpplint_FOUND TRUE)
  ament_lint_auto_find_test_dependencies()
endif()

add_executable(reading_laser src/reading_laser.cpp)
ament_target_dependencies(reading_laser rclcpp std_msgs sensor_msgs)

add_executable(grid src/Grid.cpp)
ament_target_dependencies(grid  grid_map_core)

install(TARGETS
  DESTINATION lib/${PROJECT_NAME}/
  ARCHIVE DESTINATION lib
  LIBRARY DESTINATION lib
  RUNTIME DESTINATION lib/${PROJECT_NAME}
)

ament_package()
`

samprt-heracles, 2023-05-10:
Hello, how did you fix this error ?

GradyLiugy, 2024-03-08:
Hello, how did you fix this error ?

Ryanf55, 2024-03-20:
Apparently this still affects humble. Re-opened.

StefanFabian, 2024-06-10:
I've traced it to this cmake file which pollutes the definitions of any project that `find_package`s `grid_map_core`:
https://github.com/ANYbotics/grid_map/blob/master/grid_map_core/cmake/grid_map_core-extras.cmake
here https://github.com/ANYbotics/grid_map/blob/ros2/grid_map_core/CMakeLists.txt#L130

and your `reading_laser` target does not link against `grid_map_core` so it doesn't have the include path for `grid_map_core`.

Ryanf55, 2024-06-10:
Nice find. Would you be interested to issue a PR to fix it?

StefanFabian, 2024-06-11:
Unfortunately, I do not have the time to properly fix this and check that this doesn't have side effects.
I assume it would be enough to add these definitions to the target with a public interface instead of a separate cmake.

Ryanf55, 2024-07-28:
What do you mean by `reading_laser`? I don't see that in the source tree.

Ryanf55, 2024-07-28:
Can you supply a minimum reproducible example to demonstrate the problem? CI is passing the build.

StefanFabian, 2024-07-29:
The problem will only appear in projects consuming this library since it's leaking global cmake configurations.
Simple example:

* New project with two executables.
* `find_package(grid_map_core REQUIRED)`
* `target_link_libraries` for only one of the executables
* In the code for both executables, include `Eigen/Core` and do some basic matrix addition or something

Ryanf55, 2024-07-31:
Thanks. I've pushed to a branch to reproduce this issue.
https://github.com/Ryanf55/grid_map/tree/382-bug

Here's the problem I see.
```
ryan@B650-970:~/Dev/ros2_ws/src/grid_map$ colcon build --packages-up-to issue382
[0.185s] WARNING:colcon.colcon_core.package_selection:Some selected packages are already built in one or more underlay workspaces:
        'grid_map_core' is in: /opt/ros/humble
If a package in a merged underlay workspace is overridden and it installs headers, then all packages in the overlay must sort their include directories by workspace order. Failure to do so may result in build failures or undefined behavior at run time.
If the overridden package is used by another package in any underlay, then the overriding package in the overlay must be API and ABI compatible or undefined behavior at run time may occur.

If you understand the risks and want to override a package anyways, add the following to the command line:
        --allow-overriding grid_map_core

This may be promoted to an error in a future release of colcon-override-check.
Starting >>> grid_map_cmake_helpers
Finished <<< grid_map_cmake_helpers [0.05s]
Starting >>> grid_map_core
Finished <<< grid_map_core [0.18s]                     
Starting >>> issue382
--- stderr: issue382                             
<command-line>: fatal error: grid_map_core/eigen_plugins/FunctorsPlugin.hpp: No such file or directory
compilation terminated.
gmake[2]: *** [CMakeFiles/main2.dir/build.make:76: CMakeFiles/main2.dir/main2.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:165: CMakeFiles/main2.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:91: all] Error 2
---
Failed   <<< issue382 [0.50s, exited with code 2]

Summary: 2 packages finished [0.88s]
  1 package failed: issue382
  1 package had stderr output: issue382
  ```
  
  I likely don't have time to resolve this in the next two weeks, so help is appreciated.

Ryanf55, 2024-07-31:
I tried applying patch faa4ae0c4eea364d824ccecb6e9f7a2ac0293ff4, but found that applying these definitions to the target is more complicated than initially proposed by a public target property because this file is exported.

As my branch stands now, this is the configuration error I get trying to use grid_map_core:
```
CMake Error at /home/ryan/Dev/ros2_ws/src/grid_map/install/grid_map_core/share/grid_map_core/cmake/grid_map_core-extras.cmake:7 (target_compile_definitions):
  Cannot specify compile definitions for target "grid_map_core" which is not
  built by this project.
Call Stack (most recent call first):
  /home/ryan/Dev/ros2_ws/src/grid_map/install/grid_map_core/share/grid_map_core/cmake/grid_map_coreConfig.cmake:41 (include)
  CMakeLists.txt:4 (find_package)

CMake Error at /home/ryan/Dev/ros2_ws/src/grid_map/install/grid_map_core/share/grid_map_core/cmake/grid_map_core-extras.cmake:16 (target_compile_definitions):
  Cannot specify compile definitions for target "grid_map_core" which is not
  built by this project.
Call Stack (most recent call first):
  /home/ryan/Dev/ros2_ws/src/grid_map/install/grid_map_core/share/grid_map_core/cmake/grid_map_coreConfig.cmake:41 (include)
  CMakeLists.txt:4 (find_package)
  ```

StefanFabian, 2024-07-31:
I don't think you need that config extras file at all.
Adding it to your CMakeLists should be sufficient.
The problem is that if you add such a config extras file, every project using the library will also have it added and they won't have the target for which you are adding the definitions.
Hence, the error.
You only want this to be applied to your target in your library so there is no need to export anything manually.
If set to public the compile definitions should be exported to consuming target automatically.

Ryanf55, 2024-11-28:
> I don't think you need that config extras file at all. Adding it to your CMakeLists should be sufficient. The problem is that if you add such a config extras file, every project using the library will also have it added and they won't have the target for which you are adding the definitions. Hence, the error. You only want this to be applied to your target in your library so there is no need to export anything manually. If set to public the compile definitions should be exported to consuming targets automatically.

If you have availability, can you please submit a PR.  Are you saying I change this to PUBLIC
https://github.com/ANYbotics/grid_map/pull/475/files#diff-be85320bce6ea1f2b8f51cdf6b1a569db10dd2452203ee28d096cd46fd20780fR7

But moved into the main CMakeLists instead of a confix-extras file?