ANYbotics/grid_map #390: The imported target "Qt5::Gui" references the file  "/usr/lib/x86_64-linux-gnu/libGL.so" but this file does not exist.

Inigovelasco, 2023-04-09:
Hi,

I just cloned the repository from github into a new workspace, and when I do "catkin_make" (after downloading the corresponding packages), this error appears, saying: Invoking "make cmake_check_build_system" failed. Here you can all see the whole message error:

--------------------------------------------------------------------------------------------------------------------------------------------------------
-- Using Qt5 based on the rviz_QT_VERSION: 5.12.8
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/Qt5Gui/Qt5GuiConfig.cmake:27 (message):
  The imported target "Qt5::Gui" references the file

     "/usr/lib/x86_64-linux-gnu/libGL.so"

  but this file does not exist.  Possible reasons include:

  * The file was deleted, renamed, or moved to another location.

  * An install or uninstall procedure did not complete successfully.

  * The installation package was faulty and contained

     "/usr/lib/x86_64-linux-gnu/cmake/Qt5Gui/Qt5GuiConfigExtras.cmake"

  but not all the files it references.

Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/Qt5Gui/Qt5GuiConfigExtras.cmake:45 (_qt5_Gui_check_file_exists)
  /usr/lib/x86_64-linux-gnu/cmake/Qt5Gui/Qt5GuiConfigExtras.cmake:69 (_qt5gui_find_extra_libs)
  /usr/lib/x86_64-linux-gnu/cmake/Qt5Gui/Qt5GuiConfig.cmake:186 (include)
  /usr/lib/x86_64-linux-gnu/cmake/Qt5Widgets/Qt5WidgetsConfig.cmake:101 (find_package)
  /usr/lib/x86_64-linux-gnu/cmake/Qt5/Qt5Config.cmake:28 (find_package)
  grid_map/grid_map_rviz_plugin/CMakeLists.txt:55 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/inigo/catkin_ws_thesis_grid_map/build/CMakeFiles/CMakeOutput.log".
See also "/home/inigo/catkin_ws_thesis_grid_map/build/CMakeFiles/CMakeError.log".
make: *** [Makefile:320: cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

--------------------------------------------------------------------------------------------------------------------------------------------------------
If you have any suggestions, please reach me.
Thank you!

Inigovelasco, 2023-04-20:
PROBLEM FIXED

This was because there were some incompatibilities with the Nvidia and Nouveau drivers. When above's issue turned, I just switched in the "Software & Updates" --> "Additional Drivers", from one to the other.

In my case:

I had Nouveau as default. After this issue, I turned to Nvidia, compile with catkin_make and turn again to Nouveau. Then it worked.

For the one reading this, I do not know what you did with the drivers installation, but for sure you were as lost as me. I hope this worked and you did not expend 5 days as me :)
Regards!!!

closed 2023-04-20 by Inigovelasco

lucasbrodo, 2023-11-16:
Hi,
To complete your answer, for those having problems with `libGL.so`, follow this [answer](https://github.com/colmap/colmap/issues/1271#issuecomment-926930890).