ANYbotics/grid_map #397: Build issue possibly related to Eigen add-on macros

ab3nd, 2023-08-09:
Possibly related to #256 #67 and #73 , except that #73 is supposed to have fixed this, and I'm still seeing it. 

ROS Noetic on Ubuntu 20.04.6

I am trying to write a program that, given a rosbag, iterates a topic in that bag that contains messages that contain grid_maps, and combine all of the grid maps into one map which I can then render as an image for visualization and adding illustrations to documentation. 

When I try to build, I get the error message:

<pre>In file included from <b>/my/home/dir/Code/map_render/map_render.cpp:12</b>:
<b>/opt/ros/noetic/include/grid_map_cv/GridMapCvConverter.hpp:</b> In static member function ‘<b>static bool grid_map::GridMapCvConverter::toImage(const grid_map::GridMap&amp;, const string&amp;, int, cv::Mat&amp;)</b>’:
<b>/opt/ros/noetic/include/grid_map_cv/GridMapCvConverter.hpp:177:47:</b> <font color="#CC0000"><b>error: </b></font>‘<b>const Matrix</b>’ {aka ‘<b>const class Eigen::Matrix&lt;float, -1, -1&gt;</b>’} has no member named ‘<b>minCoeffOfFinites</b>’
  177 |     const float minValue = gridMap.get(layer).<font color="#CC0000"><b>minCoeffOfFinites</b></font>();
      |                                               <font color="#CC0000"><b>^~~~~~~~~~~~~~~~~</b></font>
<b>/opt/ros/noetic/include/grid_map_cv/GridMapCvConverter.hpp:178:47:</b> <font color="#CC0000"><b>error: </b></font>‘<b>const Matrix</b>’ {aka ‘<b>const class Eigen::Matrix&lt;float, -1, -1&gt;</b>’} has no member named ‘<b>maxCoeffOfFinites</b>’
  178 |     const float maxValue = gridMap.get(layer).<font color="#CC0000"><b>maxCoeffOfFinites</b></font>();
      |                                               <font color="#CC0000"><b>^~~~~~~~~~~~~~~~~</b></font>
make[2]: *** [CMakeFiles/map_render.dir/build.make:63: CMakeFiles/map_render.dir/map_render.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/map_render.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
</pre>

I'd upload the C++ code and CMakeLists.txt, but github indicates that it doesn't support those file types. I'll try putting them in a comment to this issue.

ab3nd, 2023-08-09:
```
cmake_minimum_required(VERSION 3.16)
project(map_render)
set(CMAKE_CXX_FLAGS "-std=c++11")

set(ROS1_DIR "/opt/ros/noetic")
set(TRAVERSE_DIR "/path/to/our/custom/msgs")
add_library(
   ros1
   INTERFACE
)

target_include_directories(
   ros1
   INTERFACE
   ${ROS1_DIR}/include
   ${TRAVERSE_DIR}/include
   ${OpenCV_INCLUDE_DIRS}
)

target_link_directories(
   ros1
   INTERFACE
   ${ROS1_DIR}/lib
   ${TRAVERSE_DIR}/lib
)

# This is all used by GridMap
find_package (Eigen3 3.3 REQUIRED NO_MODULE)
find_package( OpenCV REQUIRED )

target_link_libraries(
   ros1
   INTERFACE
   roscpp
   rosbag
   rostime
   rosbag_storage
   cpp_common
   roscpp_serialization
   ${OpenCV_LIBS}
   Eigen3::Eigen
   grid_map_cv
   grid_map_core
   grid_map_ros    
)

add_executable(map_render map_render.cpp)
target_link_libraries(map_render ros1)

```

ab3nd, 2023-08-09:
```
/*
 * Takes as input a bag file, runs through the bag file and combines all grid_map messages on a given topic
 * into one grid_map, and then renders that message as an image.
 *
 * If you get errors about dynamically loaded libs, it's because you didn't source /opt/ros/(your distro)/setup.bash
 *
 **/

#include <grid_map_cv/GridMapCvConverter.hpp>
#include <rosbag/bag.h>
#include <rosbag/view.h>
#include <grid_map_msgs/GridMap.h>
#include <grid_map_core/GridMap.hpp>
#include <traversability_msgs/GridMapStamped.h>
#include <grid_map_ros/GridMapRosConverter.hpp>

int main(int argc, char **argv)
{
    // Load the bag from a file
    // TODO make bag path a command line parameter
    std::string bag_path = "/path/to/some.bag";
    rosbag::Bag bag;
    bag.open(bag_path, rosbag::bagmode::Read);

    // Can list multiple topics here
    // TODO make a command line parameter(?)
    // For now, I only care about rendering POP-T2A output
    std::vector<std::string> topics;
    topics.push_back(std::string("/featured_map"));

    // GridMaps can be combined, so create a GridMap that we can put all the
    // others into
    grid_map::GridMap full_map;
    rosbag::View view(bag, rosbag::TopicQuery(topics));
    int count = 0;
    for (rosbag::MessageInstance const m : view)
    {
        traversability_msgs::GridMapStamped::ConstPtr s = m.instantiate<traversability_msgs::GridMapStamped>();
        if (s != nullptr)
        {
            if (count == 0)
            {
                //First one, convert from the message
                grid_map::GridMapRosConverter::fromMessage(s->map, full_map);
            }
            else
            {
                //Subsequent, add to the first
                grid_map::GridMap tmp_grid_map;
                grid_map::GridMapRosConverter::fromMessage(s->map, tmp_grid_map);
                // Extend the map, don't overwrite previous data, keep all the layers
                full_map.addDataFrom(tmp_grid_map, true, false, true);
            }
        }

        count++;
    }

    std::cout << "Handled " << count << " messages." << std::endl;
    bag.close();

    // For debugging, list the layers
    for(std::string layer_name: full_map.getLayers())
    {
        std::cout << layer_name << std::endl;
    }
    //We have all the data in the grid map, now render it
    // cv::Mat image;
    // grid_map::GridMapCvConverter::toImage<uint8_t, 3>(full_map, "", CV_8UC3, image);
}
```

silencht, 2023-10-30:
```cmake
#you can try adding above this line of code
include(your_grid_map_core_path/cmake/grid_map_core-extras.cmake)
find_package (Eigen3 3.3 REQUIRED NO_MODULE)
```