ANYbotics/grid_map #399: How to convert grid map from Elevation Mapping to CostMap2D?

Aishkrish18, 2023-09-20:


deweykai, 2023-11-10:
You can use the grid_map_visualization node to convert a grid map topic into several others including an occupancy grid. Most of the grid_map demos do this in their launch file.

Also, you can include grid_map_ros/GridMapRosConvertor.hpp to convert a grid_map to other ros representations including a costmap2d.

deweykai, 2023-11-30:
Looking at it again, the `toCostmap` function is only available in the ros2 branch. You can use the `grid_map_visualization` node to output a occupancy grid like shown [here](https://github.com/ANYbotics/grid_map/blob/master/grid_map_demos/config/filters_demo.yaml) ([launch](https://github.com/ANYbotics/grid_map/blob/master/grid_map_demos/launch/filters_demo.launch)) for traversability. Then use a `costmap_2d_node` with a static layer listening to the occupancy grid topic. 

Or if you want to directly output a costmap2d. I think you can use [`Costmap2DConverter`](https://github.com/ANYbotics/grid_map/blob/master/grid_map_costmap_2d/include/grid_map_costmap_2d/Costmap2DConverter.hpp#L159). Pseudo code:

```c++
// variables
grid_map::GridMap map;
costmap_2d::Costmap2D outputMap;
grid_map::Costmap2DConverter<grid_map::GridMap> converter;

// initialize
converter.initializeFromGridMap(grid, costmap);

// create costmap
converter.setCostmap2DFromGridMap(grid, "layer name", costmap);
```