ANYbotics/grid_map #413: Request to add maintainers for ROS 2 or  transfer of ownership of ROS 2 to ethz-asl or

Ryanf55, 2024-01-20:
Hello,

I would like to request to be added as a maintainer for the ROS 2 maintenance. 

Option 1: Add me as a maintainer
Option 2: Endorse having the ROS 2 port at https://github.com/ethz-asl/grid_map, and direct users there for pull requests.
I'm open to other options.

I am currently maintaining a few related libraries in ROS 2 like grid_map_geo, which heavily depends on grid_map.

While this repository is maintained for ROS 1, the same is not as clear for ROS 2. Because ANYbotics has not demonstrated much ability to help maintain the ROS 2 port of this library. By opening up the library to more maintainers, this will alleviate some of the open issues and PR's for documentation and building the repo. Without this, it's become difficult to use grid_map in ROS 2, especially as the ROS 2 branch becomes further out of date.

Furthermore, the branch structure is not common practice for ROS 2, there is no policy or information on how to contribute to which branch. Obviously, foxy is now EOL, so those PR's may need forward-ported to a LTS release such as `humble`

For clarity, here is a list of issues affecting ROS 2 that have not been addressed:
* ros2 https://github.com/ANYbotics/grid_map/pull/245
* foxy-devel https://github.com/ANYbotics/grid_map/pull/312
* foxy https://github.com/ANYbotics/grid_map/issues/314
* ros2 https://github.com/ANYbotics/grid_map/issues/319
* foxy https://github.com/ANYbotics/grid_map/pull/341
* foxy https://github.com/ANYbotics/grid_map/issues/357
* foxy-devel https://github.com/ANYbotics/grid_map/pull/389
* foxy https://github.com/ANYbotics/grid_map/pull/394
* foxy https://github.com/ANYbotics/grid_map/issues/395
* ros2 https://github.com/ANYbotics/grid_map/issues/401
* humble https://github.com/ANYbotics/grid_map/issues/403
* rolling https://github.com/ANYbotics/grid_map/pull/405
* humble https://github.com/ANYbotics/grid_map/pull/404
* humble https://github.com/ANYbotics/grid_map/pull/407
* humble https://github.com/ANYbotics/grid_map/pull/408
* humble: https://github.com/ANYbotics/grid_map/issues/409

As part of maintaining grid_map_geo, I would take the following approach:
* start using labels to categorize issues and PR's with whether they are ROS 1 or ROS 2
* Simplify and document the branching strategy to only use `rolling`, `iron` and `humble`
* Close anything related to EOL ROS distritions that can't be reproduced in humble
* Create an issue template such that issue reporters are required to fill out a ROS distribution
* Prioritize compilation issues first, then address the reported memory leaks or compile warnings that cause undefined behavior
* Define a strategy to merge changes from `master` into ROS 2 branchs

maximilianwulf, 2024-01-31:
Thanks a lot for the request. You should be maintainer now.

Ryanf55, 2024-01-31:
Thanks! All issues in the above list now have the `ros2` labe.

Ryanf55, 2025-01-18:
Humble and Jazzy with many of the outstanding issues have been patched, and are soon to be released.
Once released, I am closing this as completed.

* https://github.com/ros/rosdistro/pull/44225
* https://github.com/ros/rosdistro/pull/44225

closed 2025-01-26 by Ryanf55