ANYbotics/grid_map #415: rolling: CI can't clone

Ryanf55, 2024-02-01:
# Bug Description

CI appears to be broken on all ROS 2 branches. No run has passed in months: 
https://app.circleci.com/pipelines/github/ANYbotics/grid_map

Lately, it can't clone

# Help requested

I don't know CircleCI. If anyone has time to lend a hand, please submit a fix and I'm happy to review it.

jtaveau, 2024-02-10:
Looking at the [logs](https://app.circleci.com/pipelines/github/ANYbotics/grid_map/431/workflows/06e21de1-5376-4043-b35d-297ca5b76dbd/jobs/415?invite=true#step-101-430_0), that's a problem with the SSH key linking GitHub and CircleCI. Is the SSH key outdated? 

Found this [doc](https://circleci.com/docs/github-apps-integration/#project-settings-ssh-keys).

Ryanf55, 2024-02-16:
> Looking at the [logs](https://app.circleci.com/pipelines/github/ANYbotics/grid_map/431/workflows/06e21de1-5376-4043-b35d-297ca5b76dbd/jobs/415?invite=true#step-101-430_0), that's a problem with the [...]

Thanks for checking in. The current maintainers have lost access to the CircleCI administration console and are unable to fix access. Instead, we are switching to GitHub actions which are already enabled and working.

Ryanf55, 2024-02-20:
CI is working now on all maintained branches. Closing as completed.

closed 2024-02-20 by Ryanf55

maximilianwulf, 2024-02-20:
@Ryanf55 great job :clap: