ANYbotics/grid_map #417: Deprecation of OgreVector3.h - Fix by ROS 2 Jazzy

Ryanf55, 2024-02-03:
# Environment

ROS 2 rolling, Ubuntu 22.04, using docker on Feb 2, 2024.

# Steps to reproduce

Place this dockerfile at the root of the repo, and build it.
```bash
docker build .
```
**Dockerfile**
```docker
ARG ROS_DISTRO=rolling
FROM ros:${ROS_DISTRO}-ros-core

RUN apt-get update \
    && apt-get install -y \
        ros-dev-tools \
        wget
        
WORKDIR /root/ros2_ws/
RUN mkdir -p src
COPY tools/ros2_dependencies.repos .
RUN vcs import --input ros2_dependencies.repos src
RUN rosdep init

COPY . src/grid_map
RUN ls src/grid_map

SHELL ["/bin/bash", "-c"]
RUN apt-get update \
    && rosdep update

RUN source /opt/ros/${ROS_DISTRO}/setup.bash \
    && rosdep install -y --ignore-src --from-paths src --skip-keys slam_toolbox

RUN source /opt/ros/${ROS_DISTRO}/setup.bash \ 
    && colcon build --symlink-install --packages-up-to grid_map
```

# Logs

```
#16 61.53 --- stderr: grid_map_rviz_plugin
#16 61.53 In file included from /root/ros2_ws/src/grid_map/grid_map_rviz_plugin/src/GridMapVisual.cpp:13:
#16 61.53 /opt/ros/rolling/opt/rviz_ogre_vendor/include/OGRE/OgreVector3.h:2:62: note: ‘#pragma message: /opt/ros/rolling/opt/rviz_ogre_vendor/include/OGRE/OgreVector3.h is deprecated, migrate to Ogre.h’
#16 61.53     2 | #pragma message( __FILE__ " is deprecated, migrate to Ogre.h")
#16 61.53       |          
```

# Target fix date

This should probably be fixed for ROS 2 Jazzy. I'm not sure when upstream will remove it, but it's still there today.

https://github.com/OGRECave/ogre/blob/master/OgreMain/include/OgreVector3.h

jtaveau, 2024-02-10:
Hi @Ryanf55, can you assign it to me? I started looking at it.

Ryanf55, 2024-02-10:
> Hi @Ryanf55, can you assign it to me? I started looking at it.

Awesome, that would be wonderful to have a hand. Please tag me in a pull request when you have this ready!

jtaveau, 2024-02-10:
I cannot push to the repo...
`ERROR: Permission to ANYbotics/grid_map.git denied to jtaveau.`

PR #422 (2024-02-12): update deprecated ogre header file
Closes #417
closed 2024-02-12 by Ryanf55