ANYbotics/grid_map #441: grid_map_pcl and grid_map_sdf warnings with COMP0144

Ryanf55, 2024-02-17:
# Background

Building `grid_map_pcl` and `grid_map_sdf` shows these warnings related to how PCL and FLANN are found. This needs to be fixed by Jazzy which comes with CMake 3.28.

# Environement

* ROS 2 `rolling`
* CMake 3.28 
* Ubuntu 22.04

# Logs

```
CMake Warning (dev) at /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:146 (find_package):
  Policy CMP0144 is not set: find_package uses upper-case <PACKAGENAME>_ROOT
  variables.  Run "cmake --help-policy CMP0144" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  CMake variable EIGEN_ROOT is set to:

    /usr/include/eigen3

  For compatibility, find_package is ignoring the variable, but code in a
  .cmake module might still use it.
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:299 (find_eigen)
  /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:543 (find_external_library)
  CMakeLists.txt:10 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /usr/lib/x86_64-linux-gnu/cmake/pcl/Modules/FindFLANN.cmake:44 (find_package):
  Policy CMP0144 is not set: find_package uses upper-case <PACKAGENAME>_ROOT
  variables.  Run "cmake --help-policy CMP0144" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  CMake variable FLANN_ROOT is set to:

    /usr

  For compatibility, find_package is ignoring the variable, but code in a
  .cmake module might still use it.
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:256 (find_package)
  /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:301 (find_flann)
  /usr/lib/x86_64-linux-gnu/cmake/pcl/PCLConfig.cmake:543 (find_external_library)
  CMakeLists.txt:10 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

** WARNING ** io features related to pcap will be disabled

```

jtaveau, 2024-02-20:
Hey @Ryanf55, I can check this one.

Ryanf55, 2024-02-20:
Help would be appreciated as long as you can get it done soon (this week). I'm hoping to release 2.2.0 by the end of the week as pretty much all the other issues are done.

Ryanf55, 2024-11-28:
Anyone is welcome to fix this.