ANYbotics/grid_map #451: There is a problem in grid_map_pcl config

juanvon, 2024-03-10:
[image]
Here the pcd path can reconfig correctly.
But the configfilepath cannot change the path.
It seems that the code need “config_file_path” but the arg name in launch is "configFilePath_", so the path to yaml is cannot change as pcd path method.
[image]

Little problem

Ryanf55, 2024-03-11:
Can you upload which version of ROS you are using this with and how you installed grid_map?

juanvon, 2024-03-11:
ROS noetic
installed grid_map from source code, master branch

grid_map/grid_map_pcl/src/helpers.cpp  line : 47 
it needs "config_file_path" but the param in launch named "configFilePath_" which in grid_map_pcl_loader_node.launch at line: 21